EdgeTX/edgetx #1624: The wizard does not work correctly on Frsky X10

Skippydi, 2022-02-21:
The wizard does not work correctly under Edgetx 2.6. New models cannot be created. Transmitter is a Frsky X10

**General information**
 - EdgeTX 2.6.0
 - Hardware Frsky X10s

pfeerick, 2022-02-23:
When you say "new models cannot be created"... What exactly do you mean? As the wizards have never actually created a new model, only edited the current one (i.e. blank model is created, then wizard run to edit it).

Skippydi, 2022-02-23:
Hi, thanks for the fast reply. 
Ok, I create an empty model to AFTER start the wizard to set the channel assignment??
What is the purpose of that?
In OpenTX I can create a finished model directly WITH the wizard, why should it be any different with EdgeTX?

closed 2022-02-23 by Skippydi