EdgeTX/edgetx #1628: (NV14) OUTPUTS / ADD all trims to Subtrims

Outwide, 2022-02-22:
In OUTPUTS / ADD all trims to Subtrims channel 1 seem ok but if you try to set other channels for the negative field (-100)
you can change it up 1 then it jumps to the name field and it can crash in either the negative field (-100) or name field area. this is where you get the fault crashes and enter emergency mode or freezes as you try to get it back . You may cannot turn off so remove batteries.
Also not easy or possible to type in the name field at top as other character appear, cannot clear the field to replace.

Expected behavior
To be able to type on the field selected and make changes easily.

General information
Edgetx 2.7
Edgetx-firmware-nightly-67d4a29e 
FLYSKY NV14

Xavitorradas, 2022-02-27:
It also happens with TX16S firmware.

Outwide, 2022-02-27:
Hi good to hear from you. I have also noticed high battery usage when it crashes and when charging the batteries in the transmitter and try the check how much recharging has been done it it show on the screen, emergency mode. I ask have noticed this aswell.

Xavitorradas, 2022-02-27:
There are some else issues but i did not take note of them. I'm looking for them. As soon as I get them I'll write.
Nice to hear you too.

eshifri, 2022-03-08:
I think this must be fixed by #1657

pfeerick, 2022-03-08:
Yeah, I think so... I got tricked by the mention of "ADD all trims to Subtrims". Thanks ;)

pfeerick, 2022-03-08:
btw, long press of enter will give a 'clear' option

closed 2022-03-08 by pfeerick