EdgeTX/edgetx #1629: (NV14) VRA switch top left (S1)

Outwide, 2022-02-22:
VRA switch top left (S1) when assigned in MIXES, Adjustment is too large, example; Small movement has a big adjustment and is not possible to get fine adjust is the usual way. VRB switch top Right (S2) may also need looking at as well.

The original Opentx no problem.

General information
Edgetx 2.7
Edgetx-firmware-nightly-67d4a29e 
FLYSKY NV14

eshifri, 2022-02-22:
Can you please check calibration of the VRA?

Outwide, 2022-02-23:
> Specifically, go into Radio Settings -> Hardware, hit Calibration, Centre both your sticks and the two pots (VRA, VRB) when prompted to do so, and then move both the sticks AND the pots full range when prompted to do so.

Hi peter I have just done this and it fix to S1 and S2 pots.

Thanks Geoff

Brisbane.

Outwide, 2022-02-23:
Hi eshifri I have just done this and it fix to S1 and S2 pots.

Thanks Geoff

pfeerick, 2022-02-23:
Excellent! :) And lol... I'm up the coast from you... stay dry! :)

closed 2022-02-23 by pfeerick