EdgeTX/edgetx #1631: [Zorro] key events constants missing

JimB40, 2022-02-22:
Seems like EVT_xxx_FIRST, EVT_xxx_REPT, EVT_xxx_LONG, EVT_xxx_BREAK are missing for SYS,MDL, PAGE>, PAGE<. TELEM & EXIT 

[image]

[image]

**General information**
 - EdgeTX version 2.6.0
  - Hardware RM Zorro

raphaelcoeffic, 2022-10-16:
@JimB40 can you please re-test on 2.8.0-rc3?

JimB40, 2022-10-17:
Same. Tested RC3 on radio & RC2 on simu.

Here is log from LUADEV

```
Defined KEY EVENTS for radio model: zorro

EVT_VIRTUAL_XXX events
---------------
EVT_VIRTUAL_PREV      	4099
EVT_VIRTUAL_NEXT      	4100
EVT_VIRTUAL_PREV_REPT 	nil
EVT_VIRTUAL_NEXT_REPT 	nil
EVT_VIRTUAL_DEC       	4099
EVT_VIRTUAL_INC       	4100
EVT_VIRTUAL_DEC_REPT  	nil
EVT_VIRTUAL_INC_REPT  	nil
EVT_VIRTUAL_PREV_PAGE 	34
EVT_VIRTUAL_NEXT_PAGE 	35
EVT_VIRTUAL_MENU 	37
EVT_VIRTUAL_MENU_LONG 	133
EVT_VIRTUAL_ENTER 	33
EVT_VIRTUAL_ENTER_LONG 	129
EVT_VIRTUAL_EXIT 	32

EVT_XXX events
----------------------
EVT_PAGE_FIRST 	nil
EVT_PAGE_LONG 	nil
EVT_PAGE_REPT 	nil
EVT_PAGE_BREAK 	nil
EVT_MENU_FIRST 	nil
EVT_MENU_LONG 	nil
EVT_MENU_REPT 	nil
EVT_MENU_BREAK 	nil
EVT_ENTER_FIRST 	97
EVT_ENTER_LONG 	129
EVT_ENTER_REPT 	65
EVT_ENTER_BREAK 	33
EVT_EXIT_FIRST 	nil
EVT_EXIT_LONG 	nil
EVT_EXIT_REPT 	nil
EVT_EXIT_BREAK 	32
EVT_PLUS_FIRST 	nil
EVT_PLUS_LONG 	nil
EVT_PLUS_REPT 	nil
EVT_PLUS_BREAK 	nil
EVT_MINUS_FIRST 	nil
EVT_MINUS_LONG 	nil
EVT_MINUS_REPT 	nil
EVT_MINUS_BREAK 	nil
EVT_ROT_LEFT 	4099
EVT_ROT_RIGHT 	4100
EVT_UP_FIRST 	nil
EVT_UP_LONG 	nil
EVT_UP_REPT 	nil
EVT_UP_BREAK 	nil
EVT_DOWN_FIRST 	104
EVT_DOWN_LONG 	136
EVT_DOWN_REPT 	72
EVT_DOWN_BREAK 	40
EVT_RIGHT_FIRST 	nil
EVT_RIGHT_LONG 	nil
EVT_RIGHT_REPT 	nil
EVT_RIGHT_BREAK 	nil
EVT_LEFT_FIRST 	nil
EVT_LEFT_LONG 	nil
EVT_LEFT_REPT 	nil
EVT_LEFT_BREAK 	nil
EVT_SHIFT_FIRST 	nil
EVT_SHIFT_LONG 	nil
EVT_SHIFT_REPT 	nil
EVT_SHIFT_BREAK 	nil
EVT_MODEL_FIRST 	nil
EVT_MODEL_LONG 	nil
EVT_MODEL_REPT 	nil
EVT_MODEL_BREAK 	nil
EVT_SYS_FIRST 	nil
EVT_SYS_LONG 	nil
EVT_SYS_REPT 	nil
EVT_SYS_BREAK 	nil
EVT_TELEM_FIRST 	nil
EVT_TELEM_LONG 	nil
EVT_TELEM_REPT 	nil
EVT_TELEM_BREAK 	nil
EVT_PAGEUP_FIRST 	nil
EVT_PAGEUP_LONG 	nil
EVT_PAGEUP_REPT 	nil
EVT_PAGEUP_BREAK 	nil
EVT_PAGEDN_FIRST 	nil
EVT_PAGEDN_LONG 	nil
EVT_PAGEDN_REPT 	nil
EVT_PAGEDN_BREAK 	nil

```

I don't care much as I wrote long ago in LUA module (database) that translates event number to key action name checking model name first.
Concept with constants was nice to have cross-compatibity using names, but number of bugs pushed me to make own implementation.

JimB40, 2022-10-17:
@raphaelcoeffic if you touch Zorro this one is quite annoying and asked by users to fix
https://github.com/EdgeTX/edgetx/issues/1888
as ToolBox configuration module doesn't work on Zorro when script is run as telemetey (permanent) mode.
PAGE key events are not passed.

lshems, 2022-11-01:
The above is true for the standalone script run, to test with:

```
local toolName = "TNS|TrimMode|TNE"
local lastEvent = ""
local eventTime = 0

-- String identifying events
local function evt2str(event)
  if event == EVT_VIRTUAL_PREV then return "EVT_VIRTUAL_PREV"
  elseif event == EVT_VIRTUAL_NEXT then return "EVT_VIRTUAL_NEXT"
  elseif event == EVT_VIRTUAL_DEC then return "EVT_VIRTUAL_DEC"
  elseif event == EVT_VIRTUAL_INC then return "EVT_VIRTUAL_INC"
  elseif event == EVT_VIRTUAL_PREV_PAGE then return "EVT_VIRTUAL_PREV_PAGE"
  elseif event == EVT_VIRTUAL_NEXT_PAGE then return "EVT_VIRTUAL_NEXT_PAGE"
  elseif event == EVT_VIRTUAL_MENU then return "EVT_VIRTUAL_MENU"
  elseif event == EVT_VIRTUAL_ENTER then return "EVT_VIRTUAL_ENTER"
  elseif event == EVT_VIRTUAL_MENU_LONG then return "EVT_VIRTUAL_MENU_LONG"
  elseif event == EVT_VIRTUAL_ENTER_LONG then return "EVT_VIRTUAL_ENTER_LONG"
  elseif event == EVT_VIRTUAL_EXIT then return "EVT_VIRTUAL_EXIT"
  else 
    local txt = string.format("Event = %i", event)
    return txt
  end
end

local function run(event)
  
  lcd.clear()

  if event ~= 0 then -- We got a new event
     -- Save the event for subsequent cycles and mark the time
     lastEvent = evt2str(event)
     eventTime = getTime()
     -- show it in debug
     print("event, eventName ",event,lastEvent)
  end

  -- Show the last event for 1 sec. in the upper left corner
  if getTime() - eventTime < 100 then
    lcd.drawText(3, 3, lastEvent)
  end

  return 0
end
```

raphaelcoeffic, 2022-11-01:
`EVT_VIRTUAL_EXIT_LONG` is reserved for killing the script.

> shifting from full size widget to standard widget 

Could we please stay on the focus of the issue? I believe this statement is related to color radios. The original issue is about `RM Zorro`, which has a standard LCD.

raphaelcoeffic, 2022-11-01:
> Perhaps log a separate issue then?

Yes, please!