EdgeTX/edgetx #1632: Zorro FrSkyRX Clone Not Working

jmxp69, 2022-02-23:
Zorro with 2.6.0 
TX16s-1 with 2.6.0
TX16s-2 with 2.6.0
TX16s-3 with 2.6.0

All MPM 1.3.3.7 AETR

TX16s-1 = Source. FrSkyX2 D16. RX#15  -  Zorro = FrSkyRX/Clone RX#15 = Bind on both radios. TX16s-1 stops bind beeping. Zorro does not. Switch Protocol to FrSkyX2/Cloned. No bind with Rx.
TX16s-1 = Source. FrSkyX2 D16. RX#15  - TX16s-2 = FrSkyRX/Clone RX#15 = Bind on both radios. Binds instantly. Switch Protocol to FrSkyX2/Cloned. Instant bind with Rx. Confirmed control on Rx.
TX16s-1 = Source. FrSkyX2 D16. RX#15  - TX16s-3 = FrSkyRX/Clone RX#15 = Bind on both radios. Binds instantly. Switch Protocol to FrSkyX2/Cloned. Instant bind with Rx. Confirmed control on Rx.
Zorro = Source. FrSkyX2 D16. RX#15  -  TX16s-1/2/3 = FrSkyRX/Clone RX#15 = Bind on both radios. Zorro stops bind beeping. TX16x does not. Switch Protocol to FrSkyX2/Cloned. No bind with Rx.

Not sure if this is an MPM issue or Edge issue. Bottom line I can Clone from TX16s to another TX16s with no issue. Works as advertised. Cloning with the Zorro fails when Zorro is source and destination.

CaleyD, 2022-02-26:
I experienced this as well and was unable to clone on zorro

raphaelcoeffic, 2022-03-22:
@jmxp69 @CaleyD just a wild guess, but maybe the 2 radio modules are too far apart in terms of center frequency. Maybe calibrating the frequency with an original receiver could help.

gagarinlg, 2023-03-18:
stale

closed 2023-03-18 by gagarinlg