EdgeTX/edgetx #1634: Buddy.EdgeTX.flasher

JimBobWorl, 2022-02-24:
While trying to use Buddy.EdgeTX.flasher when trying to add a new device, instead of seeing STM32, I get USB2.0-CRW as the only device to add. The application does not let me proceed any further. Jim

pfeerick, 2022-02-24:
So are you connecting your radio (what model?) with the power turned off, so it connects in DFU mode? As the STM32 bootloader is only active then. 

@freshollie FYI

JimBobWorl, 2022-02-25:
Evening Peter,

Thank you for your quick response. Greatly appreciated!
I am trying to update my Radiomaster TX16s from OpenTX to Edge TX with a
new SD card keeping my OpenTx configuration on the original SD Card.
Yes, I make sure the power is off, then plug in my USB cable, then run
buddy.edgetx.org/#/flash?version=v2.6.0&target=tx16s.
At the second screen I get the following message. The only option at this
point is Go Back.
USB2.0-CRW
*20100201396000000*
*0x0BDA:0x0129*

I am following the RC Video Reviews Youtube video. "How to Install Edge TX
via Web Browser.

Any idea what may be going wrong??

Jim W.

pfeerick, 2022-02-26:
So just a double check - you're using a Chrome based web browser? And when you click the 'Add new device' button it isn't listing STM32 Bootloader as an option? If you're using the top USB on the TX16S, and a USB cable and USB port that you know works normally (i.e. when powering up the radio in bootloader (not DFU) mode the two drive letters are enumerated), I'm not sure what else you can try other than something like the ImpulseRC DFU driver fixer... https://impulserc.com/pages/downloads

[image]

JimBobWorl, 2022-02-27:
Evening Peter!

That worked using the DFU fixer. I was able to flash my TX16s with EdgeTX!
Yea!

Thank you again for your help Mr. Wizard!

Jim

pfeerick, 2022-02-27:
Fantastic! :)

closed 2022-02-27 by pfeerick