EdgeTX/edgetx #1635: Add MIN as source

kristjanbjarni, 2022-02-24:
Currently in EdgeTX if you want a fixed value for a source input you can use MAX as source if need +100 but if want -100 there is currently no source for that available.

So I propose a MIN source like MAX expect with constant value of -100

There is currently no alternate except to use some other source that happens to have a value of -100, but that is sometimes not guaranted, especially with switches.

eshifri, 2022-02-24:
Can't yo use weight -100 to achieve this?

kristjanbjarni, 2022-02-24:
> Can't yo use weight -100 to achieve this?

There is no weight when selecting value source in "Global functions"

eshifri, 2022-02-24:
So the problem is with GF only?
How about creating an Input with MAX as a source and weight -100?

pfeerick, 2022-02-24:
I believe this is specific to GF/SF Adjust GV by Source? Although a complementary MIN to the MAX is probably useful generally.

PR #3639 (2023-08-28): feature:  MIN as source
Resolves #1635

Summary of changes:
- added MIN (-MAX -> -1024) as source
closed 2023-08-28 by pfeerick