EdgeTX/edgetx #1637: Bring back "screen brightness completely off" function to color screens radios

L1cardo, 2022-02-26:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature.
You can add your comment there, or react so that we know know much interest there is in the feature request.*

**Is your feature request related to a problem? Please describe.**
I use global function to turn off my screen depending on arm or not when I am flying the quad, because I don't need to look at the screen. After upgrading to 2.6.0, the screen brightness can't be set completely off, there seems to be a lowest brightness.

**Describe the solution you'd like**
Bring back "screen brightness completely off" function to color screens radios.

**Describe alternatives you've considered**
Not alternatives yet.

**Additional context**
Add any other context or screenshots about the feature request here.

pfeerick, 2022-02-26:
What about using the auto off to set if fully off when not using any of the keys?

L1cardo, 2022-02-26:
> What about using the auto off to set if fully off when not using any of the keys?

This could do the work, but if I just want to check some data on the screen after I landed(in my case: disarm=screen on), then I have to press some button to turn the screen on.

pfeerick, 2022-02-26:
Yes, but it is still "an alternative" 😉 

This is something that needs to be improved on, but was not done accidentally, as the issue of the backlight sometimes somehow being set to 0 was more problematic as no backlight == no display with colorlcd.

L1cardo, 2022-02-26:
Yeah, I think I have to set it to "off when not using any of the keys" for now.
Hope there will be more proper way in the future.
Thanks.

pino151, 2022-02-28:
For what it's worth I would definitely want this feature back as well. At races we tend to be waiting to fly for long periods of time throughout the day, and having the lcd off would greatly improve battery life.
It's quite annoying this feature was changed, and I almost want to use an older firmware due to this

MervDale, 2022-03-01:
> > What about using the auto off to set if fully off when not using any of the keys?
> 
> In some ways this could do the work, but if I just want to check some data on the screen after I landed(in my case: disarm=screen on), then I have to press some button to turn the screen on.

I can answer this for you. the screen and especially the timer were set to a minimum for a reason  eg; if you have the screen off brightness set to 0 and the timer set to 0 you can run into the issue of not being able to recover the screen at all.  A work around all your issues is now quite effective set your screen off brightness to minimum and your timer to minimum(5 seconds) and set to Keys and the only thing that will bring the screen back is the Enter Key and or a touch on the screen itself, work all your other controls to your hearts content and the screen will stay off.... AND  the first touch of the screen to bring it back on,  when it is in this mode does not effect anything (you get on touch for free ;-Q )

kristjanbjarni, 2022-03-01:
This is something I am also interrested in getting back. I understand that this can be a major problem if you make some mistake in the settings and set brightness to 0 and either have the timer set to 0 or have the backlight value accidently on some channel that is fixed to zero, in that case you basically have just lost the screen and means of fixing the issue.

Having only 5 sec as minimum for backlight timer is a good but I think there could be a solution for the off backlight and that would be to override the off with minimum brightness when any of the system keys are pressed, this way you could have the screen off when flying and using the controls and switches but pressing any system key restores minimum brightness for some specified time before turning off again. This way off backlight is available but locking yourself out of the screen can't happen.

og54, 2022-03-02:
For me it's not really a problem that the potentiometers can't turn off the screen completely, but I would  like the OFF option returned for switch functions. That way people can either use logical switches to set the lower position of the pot to trigger an off condition, or use an actual switch in order to turn off the backlight.

Lexiyntax, 2022-03-25:
Not being able to turn this off completely is particularly annoying in night flying conditions where the mosquitos love even the lowest brightness setting. I don't think the 'lockout' issue is as big of a deal as it is made out to be either. One can just use a flashlight or other light source to 'see' the screen even with the backlight entirely off. Simplest way to fix this IMO is a 'BacklightOff' special function. This would override the system menu so for night flying models it can be turned off with a switch without affecting other models by requiring a system menu change. Also with it bound to a switch during model programming it's easy to simply toggle the switch to get the screen back.

gagarinlg, 2022-03-25:
> I don't think the 'lockout' issue is as big of a deal as it is made out to be either. One can just use a flashlight or other light source to 'see' the screen even with the backlight entirely off.

That is not true for all radio. NV14 switches the display completely off, with absolutely no image

pfeerick, 2022-03-26:
That is actually a good point... a specific BlacklightOff SF/GF... 

As part of 2.8 I would like to see the volume and brightness via pot/switch overhauled... but to do so we will probably have to a have a tracking issue for each, as the requirements will need to be very clearly set out so we don't nuke any features/capabilities that currently exist which people want. If, for instance, we rip out the setting of the brightness via SF/GF and move it to the brightness settings, we could also allow for switch override of pot, so the two controls work in tandem/mutually exclusive or or or neither get used. I'll draft up a discussion topic for each in the next couple of days outlining my thoughts on this and we can hopefully start the ball rolling from there...

MervDale, 2022-11-08:
It shouldn't be too hard to set your backlight to on all the time only when you throttle is active and back light timer of say, 5 sec becomes active when you throttle safety switch is active, I assume you would have a throttle safety switch ?

azaz44, 2022-11-08:
> It shouldn't be too hard to set your backlight to on all the time only when you throttle is active and back light timer of say, 5 sec becomes active when you throttle safety switch is active, I assume you would have a throttle safety switch ?

But how?
As it is now, if I add SF to set backlight to on (for example base on throttle), it is ignored if in radio settings backlight is set to "Keys". "Keys" (with timer) seems to have precedence.

Shonkonray, 2023-08-04:
> I also miss more backlight control. I can't do what I want with what is provided. Some examples of what I'm unable to achieve (flying rc helis):
> 
>     * I'd like the screen to stay on all the time, when I'm flying, but go off (completely) after some inactivity time when I'm not flying
> 
>     * I'd like to keep the screen off during flight, for some models (like simulator)
> 
> 
> It looks like natural way to solve this would be:
> 
>     * set backlight to "Keys" and some desired delay (like 15 s)
> 
>     * add a special function, which turns on the backlight in specific condition (for example when throttle is on, or telemetry is available) per model
> 
> 
> But for some weird reason this does not work in EdgeTX. If you set backlight to "Keys" it seems to override special functions and will turn backlight off always, even when SF sets it to ON.

I don’t know when this issue will be addressed, but in the meantime, I am using a LUA script to manage the backlight.

This is how it works:
1) Set the backlight to keys with desired off delay (in my case as short as possible, so 5s)
2) Set a GF or SF to a pot (e.g. S1) and regulate brightness (or a switch if simply on/off is enough)
3) use a LUA script and the lcd.resetBacklightTimeout() function to reset the Timeout if said pot is above desired value.

This way the backlight is always on as long pot is above threshold and switches off after 5s if its below.
Only caveat is that this only works if the widget is visible (no menu opened, or any other widget is in full screen etc.) So somewhere at the top bar is best.

In case someone is interested, then text me and i will modify the script to what you need (if possible).

MervDale, 2023-08-05:
Some help  here, maybe 

Most of what you want can be done easily, set your backlight to CONTROL and INACTIVITY TIMEOUT to what ever you desire and the backlight will be ON all the time when you are using any of your controls( FLYING) and OFF when not using any controls.
set your ON BRIGHTNESS to the max you want and control it via GF S2 in my case and OFF BRIGHTNESS to minimum It should be easy enought to set IF THEN otion in LOGICAL SWITCHES to, say one or more of the 6P buttons .
I use this option on 6P6 to switch my momentary switch(SH) to activate  TIMER RESET or to initaliase INSTRM

Shonkonray, 2023-08-05:
@MervDale thank you for your reply. but that is not what everybody does want here. 😄 
We want to have it completely OFF while flying (on SIM or night flying eg.). So the stick option you mentioned is a no go.
Also the OFF Brightness on some radios is still very very bright (like on my X10s) (due to crappy hardware).
So for now, the above LUA was the only working solution i could find, but it is still just a shabby workaround and i am sure that at some point all this can be done nicely with standard ETX. Until then, i will keep using the LUA. 😄

raphaelcoeffic, 2023-08-06:
> We want to have it completely OFF while flying

These are the settings I use for exactly this purpose:
[image]

In case this was not clear to everyone, when the screen is OFF (0% backlight), we have the following:
- key input is inactive
- touch input is inactive
- key and touch input reactivates backlight (basically resets the timeout)

If we would not enforce a minimum backlight in certain conditions, we would have no way to prevent being locked out of any UI control on the radio. You will notice that your phone works the same, so everyone should be used to similar mechanisms.

MervDale, 2023-08-06:
SHONKONRAY   So lets get things right here first as to what you really want? 
Quote "I'd like the screen to stay on all the time, when I'm flying, but go off (completely) after some inactivity time when I'm not flying"

The Contol option in Mode will do that and what ever time delay you set will switch the screen off when not using the controls with the OFF Brightness set to zero.

Quote" I'd like to keep the screen off during flight, for some models (like simulator)"

So as you said the keys option will do that 

So we also seem to have a bit of a quandry in exactly what you want.

Quote"but that is not what everybody does want here. 😄
We want to have it completely OFF while flying (on SIM or night flying eg.). So the stick option you mentioned is a no go."

Set to KEYS does work in this case and is what I use to turn my screen off when I am flying, with a touch to any key that will activte the screen if needs be.

One of the problems that came up in the early days of ETX is that if you set  OFF Brightness to 0 and your Inactivity Timeout to 0 you got stuck in a loop that was very difficult to get out of. so the OFF option and 0 seconds was removed from ETX . Still active  in OPENTX  so a trap for the unwary.

raphaelcoeffic, 2023-08-06:
> I'd like to keep the screen off during flight, for some models (like simulator)

We cannot have it both ways. Either `Sticks`, `Keys` or `Both`, but it's global, meaning that you cannot have it differently based on model.

commit db410eb2 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit 8798b51e (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit ba0fb945 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit ab3cb457 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53