EdgeTX/edgetx #1646: Update to 2.6.0 on TX16s, Internal RF doesn't recognize internal module

oleg-sherman, 2022-02-28:
### Issue
After updating to EdgeTX 2.6.0, in the model setup menu -> **Internal** RF mode, has only `OFF` option.

### Expected 
Model setup menu -> internal RF mode, should have `OFF` and `MULTI` options.

### Workaround 
Downgrade to EdgeTX 2.5.0 solves the issue.

### Notes
- Updating internal module firmware did't help.
- **External** RF seems to work fine.
- Used Radiomaster TX16s Max

pfeerick, 2022-02-28:
See if this fixes it:

https://github.com/EdgeTX/edgetx/discussions/1619#discussioncomment-2232886

oleg-sherman, 2022-02-28:
@pfeerick 
> Can you see if this fixes it:
> 
> [#1619 (comment)](https://github.com/EdgeTX/edgetx/discussions/1619#discussioncomment-2232886)

This do not fix the issue, EdgeTX companion 2.6 also missing the `MULTI` option:
[image]

pfeerick, 2022-02-28:
So when you went to the Radio Settings -> Hardware tab, and went down the "Internal module" as shown in that video, what options did you get? 

Until you can change that setting from OFF to MULTI, the model internal RF setting will not let you pick anything other than... well, OFF. 

For the equivalent setting in Companion (which appeared to work for the other person) you want:

[image]

[image]

oleg-sherman, 2022-02-28:
@pfeerick 
> So when you went to the Radio Settings -> Hardware tab, and went down the "Internal module" 

This workaround works, Thanks!

pfeerick, 2022-02-28:
Its not really a workaround - as that setting, introduced in 2.6, is what governs what module is used as the internal module - i.e. if you modify the transmitter, you would change that option. 

When you went to the Radio Settings -> Hardware tab, and went down the "Internal module" on the transmitter as shown in that video, what options did you get?

gagarinlg, 2022-06-11:
I think this can be closed, right?

closed 2022-06-11 by oleg-sherman