EdgeTX/edgetx #1647: Spectrum Analyzer not found

TheIsotopes, 2022-02-28:
**Describe the bug**
Spectrum Analyzer not seen in tools menu

**To Reproduce**
go to tools menu 

**General information**
 - EdgeTX version 2.7.0 latest master
 - RM Zorro

pfeerick, 2022-02-28:
Is this a Zorro with MPM? Only my ELRS Zorro is working atm so I can't confirm or test this. However, spectrum analyser is not a supported feature for ELRS. Not sure why it would be missing as AFAIK factory firmware should simply be a standard build of ETX (or if it's MPM maybe they shipping with OTX?) unless it is something that has been knobbled in recent builds.

TheIsotopes, 2022-03-01:
Yes, i have the 4in1 version and it was shipping with ETX 2.6.
If i*m going back to the 2.6 release version the same issues stands. Only back to factory firmware solved the issue.

TheIsotopes, 2022-03-01:
@pfeerick small update, seems to concern only b/w .
I have now tested the last master on my tx16s, the spectral analyzer is displayed and works.

pfeerick, 2022-03-02:
As I stated before, this is because the Zorro is built with default internal RF module being CRSF, whereas the TX16S is built with the multi module. And the logic that was being used to show the internal MPM spectrum analyser was dependent on that. But with those transmitters, and several others, you now have several options as to what the internal RF module could be, which is why that logic needed to be changed for both.

 Can you give the linked PR a test if you get a chance, and ensure it is working for you with the MPM Zorro? It's working perfectly for me on TX16S /w MPM, and Zorro /w ELRS. If I change the internal module in the hardware settings from MULTI to something else the entry is correctly removed from tools now, and added (even though invalid on the ELRS Zorro) if set to MULTI.

PR #1648 (2022-03-06): fix: Make decision to show int MPM spectrum analyser entry settings dependent
Fixes #1647

Showing of the "(int) Spectrum Analyser" was determined at build time,
dependent on the default internal module, which is wrong post #1117

This now makes it so it is dependent on current internal module setting (if the radio has an internal module) for both colorlcd and B&W radios.

Tested on TX16 and Zorro (ELRS). Tool entry added with MULTI, not added with CRSF.
commit c73057ad (2022-03-06): fix: Make showing int MPM spectrum analyser entry runtime determined (#1648)

Showing of the "(int) Spectrum Analyser" determined at build time,
dependent on the default internal module, which is wrong post #1117
Fixes #1647
closed 2022-03-06 by pfeerick