EdgeTX/edgetx #1652: TX16S OUTPUTS Emergency Mode

Xavitorradas, 2022-03-02:
In OUTPUTS if you try to set channels for the negative field (-100) you can change it up (+/-) but then it jumps to the name field and appear any character not entered for you and it crash in either the negative field (-100) or name field area entering in Emergency mode or blocking transmiter so you can't turn off so must remove batteries in this last case.

Expected behavior
Change data as usual in other vertions.

General information
Edgetx 2.7
Edgetx-firmware-nightly-tx16s-2929f1a
Radiomaster TX16S
[image]
[image]
[image]

pfeerick, 2022-03-03:
Related: #1639

pfeerick, 2022-03-03:
Only seems of affect "Min" field, *not* Max, and this issue specially needs touch to trigger it - using key navigation still 'breaks' in as far as Min field loses focus after one increment, but doesn't trigger a EM.

eshifri, 2022-03-03:
Interesting: I cannot reproduce neither this, nor #1639.
For me the MIN reaches limit (-150.0 - I have extended limits) and stops there.

Xavitorradas, 2022-03-03:
> Only seems of affect "Min" field, _not_ Max, and this issue specially needs touch to trigger it - using key navigation still 'breaks' in as far as Min field loses focus after one increment, but doesn't trigger a EM (which is probably somewhat due to focus jumping to controls thus touch keyboard closure trying to cleanup wrong stuff).

Not only happens in touch. Whe trying to change Chanel 4 ( Rudder ) it is inestable. moving the roll button also focus always go to NAME and only is possible to get out bay RTN button. It only happen in key navigation and in Rudder Chanel 4. In touch always break or EM

Xavitorradas, 2022-03-03:
With last a1fe0b7 just now instaled it now runs Ok in Chanel 4 by key pad but still EM in Touch.

pfeerick, 2022-03-04:
The ***it*** I was referring to was EM - as I stated "_using key navigation still 'breaks' in as far as Min field loses focus after one increment_" ... 

Similar to what the video shows, if I open CH3 or CH4, when you try to scroll down the fields using the roller, as soon as the roller stops, focus jumps to the first field. CH2 and CH1 *seem* ok in that respect. 

CH1, if you use the roller, will jump focus to the name as soon as min has changed move than 0.1 in either direction. CH2-3 jump focus on their own. CH4 seems mostly ok, but will randomly jump focus when changing min via roller or touch. 

This is on a PR test build from yesterday. Reverting to 2.6.0 for comparison shows not sign of any of the elements of these issues, which also don't seem to have any connection to state of extended limits.

pfeerick, 2022-03-04:
https://youtu.be/FDDB_HNX350

eshifri, 2022-03-04:
Now I see _it_ . The same happens on channels 5 and 6 as well. With and without inversion.

eshifri, 2022-03-04:
I think something is wrong with ROTARY_EVENTS: it all starts with filed selection jumping all over the screen. If you play with them and manage to get control over the field selection, everything works fine after that.

PR #1657 (2022-03-06): handle events on the outputs screen (#1652)
Fixes #1652 

Summary of changes: move call to Windows::checkEvents() to after update() call.
This fixies erratic behavior of the rotary selector on the outputs screen.
Maybe fixies  #1639 as well, but I was never able to reproduce this one.
closed 2022-03-06 by pfeerick