EdgeTX/edgetx #1654: X9D 2016 - radio setup gets reset upon power off occasionally.

onecooler, 2022-03-03:
Once in a while radio setup gets reset occasionally. There are no steps identified to reproduce. About once a week I find my radio reset to defaults (like no switches assigned, radio in mod 1, sticks not calibrated) upon power on. No other issues.
FW version:
2.6.0
edgetx-x9d+
Please contact me if any other information is needed.

dennisvr, 2022-04-30:
I have the same issue
FW version:
2.7.0
edgetx-x9d+

dennisvr, 2022-05-01:
i've replaced the SD card, problem seems to be gone now.

pfeerick, 2022-05-01:
Please update with how you go after a week or so. However I suspect that will have resolved the issue, as EdgeTX 2.6.0 onwards uses the SD card for radio and model settings, so if the SD card is marginal, you'll have had intermittent issues with it.

dennisvr, 2022-05-07:
I haven't encountered the issue since replacing the SD card.

closed 2023-03-18 by gagarinlg