EdgeTX/edgetx #1659: where are the bin files per model on horus?

lshems, 2022-03-04:
Is there some documentation on the way models are stored on the radio?

I used to have a file install function to copy the correct files per model from a backup directory to the active directory of the radio. These where .bin files per model. 
I would then update the models.txt file to reflect the changes.

This doesn't seem to work anymore?

eshifri, 2022-03-04:
Please take a look at the release notes.
ETX 2.6 uses YAML files instead of binary. models.yml instead of models.txt

lshems, 2022-03-04:
Yeah, I read that. But on the simulator all becomes very confusing.

I can open and save otx files, restore bin files on BW radios,...

pfeerick, 2022-03-05:
but horus != B&W, so you're comparing apples and lemons. B&W used to use EEPROM for model settings, so there weren't any model bins on SD, unless you backed up to SD. Now it uses SD card and yaml files, as do colorlcd radios.

lshems, 2022-03-05:
But I am talking about the backups. They are still bin on BW (there are no
'backups' on colour)? Will that change to yml?

In the color, I can just put a model yml in the models folder and adapt the
models.txt?

pfeerick, 2022-03-05:
> But I am talking about the backups. They are still bin on BW (there are no 'backups' on colour)? Will that change to yml?

What you can do in the case of a old BIN is copy that into models, reference it in the `/MODELS/models.yml`, and the firmware will convert it to yml the first time it is accessed. 

> In the color, I can just put a model yml in the models folder and adapt the models.txt?

Yes,  you can just put a new model.yml file in, and update `/MODELS/models.yml` to reference it.

lshems, 2022-03-05:
Super. Thanks.

closed 2022-03-05 by lshems