EdgeTX/edgetx #1660: Model-Setup: Pots&Slider-Check: Auto has no function

wimalopaan, 2022-03-05:
The pots&slider check button in model setup has three values: OFF, ON, AUTO.

Setting to ON saves actual position of all selected pots and sliders.

Setting to AUTO is no handled in source code (not even in B&W) radios

Before removing this option: does anybody know whta the intend of AUTO was (hostorically)?.

rotorman, 2022-03-05:
Auto should remember the last position when the radio was turned off, or a model changed.

wimalopaan, 2022-03-05:
Looks like that doesn't work, at least in simulator.
Grep-ing thru the code I can't find meaningful usages of POTS_WARN_AUTO.

wimalopaan, 2022-03-05:
Mmh, in simulator that does not work, I think

wimalopaan, 2022-03-05:
Ok, I misinterpreted the function: AUTO-save does its work for all pots/sliders, that are not checked.
Is that intentional???

wimalopaan, 2022-03-05:
I think it is intentional, because it allows to mix explicit setting and auto setting.

closed 2022-03-05 by wimalopaan