EdgeTX/edgetx #1661: Font sizes and names

lshems, 2022-03-05:
For some reason the auto font size selection for widgets that I based on the font sizes of the OpenTX fonts, doesn't select the right size font anymore.

Is there some info on the current font sizes?

These are the ones I currently use:

`local fontSets = 
  {[0]=   --FontSet
    {[0]={H=20,W=13,name="DEFAULT",size=0}
    ,[MINSIZE]={H=14,W=8,name="MINSIZE",size=MINSIZE}
    ,[SMLSIZE]={H=18,W=10,name="SMLSIZE",size=SMLSIZE}
    ,[MIDSIZE]={H=28,W=18,name="MIDSIZE",size=MIDSIZE}
    ,[DBLSIZE]={H=34,W=26,name="DBLSIZE",size=DBLSIZE}
    ,[XXLSIZE]={H=64,W=40,name="XXLSIZE",size=XXLSIZE}
--    {[0]={H=18,W=11,name="DEFAULT",size=0}
--    ,[MINSIZE]={H=12,W=6,name="MINSIZE",size=MINSIZE}
--    ,[SMLSIZE]={H=16,W=9,name="SMLSIZE",size=SMLSIZE}
--    ,[MIDSIZE]={H=24,W=14,name="MIDSIZE",size=MIDSIZE}
--    ,[DBLSIZE]={H=34,W=22,name="DBLSIZE",size=DBLSIZE}
--    ,[XXLSIZE]={H=64,W=40,name="XXLSIZE",size=XXLSIZE}
    }
  ,[1]= 
    {[0]={H=8,W=6,name="DEFAULT",size=0}
    ,[MINSIZE]={H=6,W=4,name="MINSIZE",size=MINSIZE}
    ,[SMLSIZE]={H=7,W=5,name="SMLSIZE",size=SMLSIZE}
    ,[MIDSIZE]={H=12,W=8,name="MIDSIZE",size=MIDSIZE}
    ,[DBLSIZE]={H=16,W=10,name="DBLSIZE",size=DBLSIZE}
    ,[XXLSIZE]={H=38,W=21,name="XXLSIZE",size=XXLSIZE}
    }
  }
  
fontSets[2] = fontSets[1]`

Then I use something like this to determine which size I should remap based on the widget size:

`local order =   {MINSIZE,SMLSIZE,0,MIDSIZE,DBLSIZE,XXLSIZE}

local fontMap = {}
local newSize = {} 

for size,value in pairs(fontSets[ZONE.display]) do
  newSize = {w=1,h=1}
  for o,s in ipairs(order) do
    if value.W/ZONE.size.w >= fontSets[LCD.radio][s].W/ZONE.zone.w then
      newSize.w = o
    end
    if value.H/ZONE.size.h >= fontSets[LCD.radio][s].H/ZONE.zone.h then
      newSize.h = o
    end
  end
  fontMap[size] = fontSets[LCD.radio][order[math.min(newSize.w,newSize.h)] ]
  fontMap[size].w=fontMap[size].W*(((ZONE.smartMode == 1) and ZONE.size.w/ZONE.zone.w) or 1)
  fontMap[size].h=fontMap[size].H*(((ZONE.smartMode == 1) and ZONE.size.h/ZONE.zone.h) or 1)
end 
`

lshems, 2022-03-05:
PS, can't ghet the code formatted correctly :(

lshems, 2022-03-05:
OK, just figured out that the MINSIZE is not available on Horus.

Can anyone enlighten me on the HORUS fonts?

Thanks!!

pfeerick, 2022-03-06:
> PS, can't get the code formatted correctly in the post above :(

Markdown uses triple backticks for multi-line blocks ;)

gagarinlg, 2023-03-18:
the smallest font is called TINSIZE, not MINSIZE
it is the same on OpenTX
https://github.com/opentx/opentx/blob/1f8d4b2b39decbba6affaa8ee2a382ec0efb599f/radio/src/lua/api_general.cpp#L1990

lshems, 2023-03-18:
Interesting. I really wonder if it changed somewhere, as I wrote this for opentx ages ago...

Thanks! I'll check if this solves my font scaling issues...

gagarinlg, 2023-03-18:
> Interesting. I really wonder if it changed somewhere, as I wrote this for opentx ages ago...

It looks like this has been introduced 3-4 years ago in OpenTX, with that name.

Eldenroot, 2023-03-18:
It should be renamed... :)

gagarinlg, 2023-03-18:
At least weekly. 😜

lshems, 2023-03-18:
Lol. Sure it was existing but no constant. So I figured it out by trial and error and added the constant as local variable.

JimB40, 2023-03-19:
Was TINSIZE AFAIR (and available on color only)
As usually otx luadoc not mention it so had to check C code.