EdgeTX/edgetx #1666: Emergency Mode when trying to change name of model categories

nostromo123, 2022-03-06:
FrSky X10S Express with edgeTX 2.6

Trying to change a name in that window here constantly results in an emergency mode.

[image]
 
Can present a video if requested.

Sorry, if that issue has been reported earlier ;-)

br KH

pfeerick, 2022-03-08:
I have this suspicion it was, but it might has been part of another issue. Either way, single item issues are easier to digest ;)

pfeerick, 2022-03-08:
If you get a chance a video would be great, as I haven't been able to repo this yet.

pfeerick, 2022-03-08:
Actually, I reverted to 2.6 as I'm on a nightly atm, and it crashed while I was fiddling with it. So can you try with a nightly and see if this is indeed fixed, as it looks like it was fixed already. Also, how did you get the `_` in, as it doesn't seem to be a valid character atm.

nostromo123, 2022-03-08:
Hi, I do all my programming with companion, not on the radio. So I did it that way with the '_'.

No video necessary, as I understand!? I'll try nightly, but at the moment I am quite busy. Maybe tomorrow ....

I'll report back :-)

cu KH

nostromo123, 2022-03-08:
Nightly on the radio, and no issue anymore!

[image]

I'll close this issue. Thank you again for giving a hand.

Have a nice day.

br KH

closed 2022-03-08 by nostromo123

pfeerick, 2022-03-09:
Ah, that makes sense (Companion)... I was thinking it might have been a manual edit... so will have to look at why it's not in the list presented on the radio at some point...