EdgeTX/edgetx #1670: The 2.6 release for Taranis X9D Plus was incorrectly compiled

CamGenius, 2022-03-07:
**Describe the bug**
The internal rf module option on the radio on gives OFF or MULTI

**To Reproduce**
Flash the firmware on the FrSky X9D Plus and then turn on the internal rf module.

**Expected behavior**
The internal rf module option on the should be OFF or XJT

**Screenshots**
[image]

**General information**
 - EdgeTX version 2.6 Santa Release
 - FrSky X9D Plus

pfeerick, 2022-03-09:
My X9D+ is currently in pieces on the floor atm (after finally getting hold of a replacement shell for it), so I can't test this. Can you try opening "RADIO\radio.yml" and look for `internalModule`. It is probably set to `MODULE_TYPE_MULTIMODULE`. Try changing it to `MODULE_TYPE_XJT_PXX1`. If that does it, just need to figure out *how* it got that way... as from a quick glance at the code it should be right. :-/

CamGenius, 2022-03-13:
Thank you that seems to have fixed the issue. Maybe consider adding this to the wiki as there is no info anywhere on how to fix this problem.

closed 2022-03-13 by CamGenius