EdgeTX/edgetx #1674: Companion - show curve graph for diff and expo

pfeerick, 2022-03-09:
After #1484 introduced a curve graph for custom curves when editing the mixes, I proposed that  it would be nice to be able to see expo and diff curves also... so here is the wishlist item ;)

_Originally posted https://github.com/EdgeTX/edgetx/issues/1484#issuecomment-1061434577_

elecpower, 2023-04-05:
@pfeerick weight, offset and/or curve can be based on GVs which can change based on current flight mode and SFs and thus the curve representation on the radio can change significantly as each of these changes.
In Companion there is no current flight mode and the GVs are at settings values.
If the curve is drawn in Companion using FM0 and default values then it is very likely not going to represent how the radio will behave and this could have bad outcomes eg model in lots of pieces if the pilot does not recheck on the radio and/or simulator.
IMO if Companion cannot mirror the radio behaviour (well at least very close) then better not to try for risk of misleading. Your thoughts?

pfeerick, 2023-04-07:
Yeah, let's table this for now. I suppose at a reach it is easy to say "well, just make it so you can set the active GV and FM", but at that point, I think I would respond "just use the simulator, that's what it's there for!" while lobbing it in the "too hard for now" basket.

closed 2023-04-07 by pfeerick