EdgeTX/edgetx #1687: can't load model.yml end of map not found

JurikRistjouw, 2022-03-12:
**Describe the bug**
When importing models and settings from radio with EdgeTX Companion 2.6.0 an error occurs.

**To Reproduce**
Import models and settings from radio

**Expected behavior**
An overview of models to be edited.

**Screenshots**
[image]

**General information**
 - EdgeTX version 2.6.0
 - Frsky X9D+
 - FW: edgetx-x9d+ (2022-01-24 07:17:44) CFGW: 221

JurikRistjouw, 2022-03-13:
Not sure it is an issue as of yet, I opened the saved "old" settings from edgetx companion and wrote it to the radio. It works and I am able to read it back.  Maybe worth to mention, that the names of the models were cut of at 12 characters when I was unable to read it from radio. This is not the case anymore, the modelnames are not cutoff anymore. Maybe a conversion mishap when flashing to edgetx?

closed 2022-03-13 by JurikRistjouw

pfeerick, 2022-03-13:
This sort of issue came up often during the development of 2.6 as the yaml configuration file format was still being changed, so it was necessary to open the model on the transmitter and make some simple change to the offending model to make it re-save the settings, and in doing so it would tweak whatever was wrong or changed. That could be the case here, or there could some weird conversion gremlin as you suggest. 

Would need to know the exact order of what happened (i.e. migrated from OTX v to ETX v, or ETXv to ETXv, on radio/using compaion), etc, in order to pin that down further.

JurikRistjouw, 2022-03-13:
Thanx for replying! Always nice to know I am not the only one.

I came from opentx 2.3.9 followed the installation method with flash from tx. Empty SD card, sd card content, copied models and sounds. Then did not turn of the tx, and exited through the bootloader, so it booted into edgetx. From there see bugreport; names cutoff and not able to load settings and models from tx in edgetx.