EdgeTX/edgetx #1688: Lua - Improve (getfieldInfo(source)).name/.desc to get user friendly name from source number.

MikePRoberts, 2022-03-12:
I have a source selected from options via { "Source", SOURCE, 89 },
I cannot find a way to get from the "Source" number to a user recognisable name for in particularly telemetry sources (e.g. Alt, Vspd, RSSI, RxBt).  getfieldInfo(source).name and .desc deliver user unhelpful results in the form "telem9"/"Telemetry sensor 9".

Just getting from the source number to the text in the SOURCE list would be a good result (e.g. via getfieldInfo(source).source?).  Or am I missing something?

Thank you  Mike

eshifri, 2022-03-12:
field = getFieldInfo(id)
field.name 
is supposed to work.
At least it worked the last time I had used it.

MikePRoberts, 2022-03-12:
Thank you for your quick reply - but this is what is giving me the unhelpful names:
Source selected in options    getFieldInfo(source).name
RSSI                                       telem12
RxBt                                       telem15
VSpd                                      telem10
Alt                                          telem9
A2                                          telem8
Tmp1                                      telem11

Note this is on a TX16S radio with EdgeTx 2.6 connected to an FrSky X8R with S-port variometer.
Will be happy to do any tests that might help diagnose.
Mike

eshifri, 2022-03-13:
I cannot reproduce it the current code and my script (version of showAll) was working in 2.6 as well.
Let's start with RSSI, since it is available for majority of the RXs.
How do you find the source value?
Could you check that your source value is correct by calling getFieldInfo("name") ?
Are telemetry sensors available for your model?  output telemX hints that there are no telemetry sensors, saved for the model.

MikePRoberts, 2022-03-13:
I hope that the attached screenshot and test Luascript will clarify.  The script selects "Source" in line 3 and the screen data is written in lines 17 to 20.  The top number is the Source number and the bottom number is its value - that corresponds to the RSSI data in the Value widget top left.
I wonder if the Value Widget might give a clue to solving the problem as it has output the helpful descriptor.
Thank you.
Mike
[Test.zip](https://github.com/EdgeTX/edgetx/files/8239393/Test.zip)

eshifri, 2022-03-13:
Sorry, I do not understand this. (And I admit it may be my problem: I do not use LUA too often.)
But I would do it like this:
	-- Draw RSSI
	field = getFieldInfo("RSSI")
	val = getValue(field.id)
	lcd.drawText (x, y, field.name, font + colorFlags)
	lcd.drawText (x + xOffset, y, (val and val ~= 0) and (math.floor(val + 0.5)) or "---", font  + colorFlags)
field.id for a sensor can change every time you run sensor re-discovery.
It can also change when new mixer inputs are added to the code.
So it is preferable to get sensor by name: e.g. getValue("RSSI").
If I call getFieldInfo(286) I'm getting telem12 as a name and Telemetry sensor 12 as a descriptor as well.
This is because the SW is searching for the id==286 in a different array.
https://github.com/EdgeTX/edgetx/blob/9e6ab8459920c065a112458bb2d80926d7fa6215/radio/src/lua/api_general.cpp#L432

jfrickmann, 2022-03-13:
I can see that it somehow fell out of the documentation, but try:
function getSourceName(sourceIndex)
@param sourceIndex: integer identifying a value source as returned by `getSourceIndex(sourceName)` or the `id` field in the table returned by `getFieldInfo`.
@retval sourceName: string naming the value source as it is shown on radio menus where a source can be chosen.
@notice the source names shown on the screen are not the same as the names used by `getFieldInfo` and `getValue`. But the indices are the same, so `getValue(index)` will work with the indices used here.
@status current Introduced in 2.6

MikePRoberts, 2022-03-13:
Thank you.  getSourceName(sourceIndex) gives user recognisable source names from sourceIndex from the options SOURCE selection.
Mike

closed 2022-03-13 by MikePRoberts