EdgeTX/edgetx #1691: Throttle Warning for a Centered Throttle

bigtwin, 2022-03-13:
I moved [this issue](https://github.com/opentx/opentx/issues/7794) from OTX repository to here:

Cars and Ships are usually capable to move backwards and therefor "Zero-Gas" is at neutral position. It would be usefull to have a _Throttle Warning_ for teh throttle stick not being in center/neutral position.
I think of something just like another check box on the Setup screen like we have it for _Reverse Throttle_. If the box ist checked, a _Throtte Warning_ will occur when the throttle stick is not in centered position at model start.

pfeerick, 2022-03-13:
You wish has already been granted  ;) 

https://github.com/EdgeTX/edgetx/pull/1395

bigtwin, 2022-03-13:
Man, thx for this hint! And sorry for not recognizing this. I am totally new with EdgeTX and currently carefully approaching it ...
I guess I supposed to close this issue, right?  (beginner in Github as well)

closed 2022-03-13 by bigtwin

pfeerick, 2022-03-13:
No worries... you just had good timing. Closing is good... I was only leaving it open until you had a change to respond. :) 
This will be a feature in 2.7, or you can use a nightly if you really need it now. I would suggest waiting though as we are starting to wrap up for 2.7, so will be starting to do the release candidates soon.