EdgeTX/edgetx #1697: The "Telemetry" item in a "AND" logical switch is not correctly saved.

dududu76, 2022-03-16:
The item "Telemetry" in logical switches dropdown list, is not correctly saved by Companion.

**Steps to reproduce the behavior**
In Companion, edit a model, and go to "Logical Switched" tab.
Define a new logical switch with
- Function set to "AND" or "OR".
- V1 parameter set to "Telemetry" or "!Telemetry".
- Other fields can be left empty.

Save the .etx file ( File -> Save ).

Close the file (File -> Close).
Re-open it : the "Telemetry" V1 parameter Has disappered.

Additionnally, when pushing to the radio, "!Telemetry" parameters are transformed into "Telemetry".

**Expected behavior**
The "Telemetry" (or !Telemetry") parameter should be correctly saved.
Also, all "!Telemetry" parameters should not be turned into "Telemetry" when pushed to the radio.

**General information**
 - EdgeTX release version (2.6.0)
 - Hardware : Frsky X9D+
 - OS : Windows64

elecpower, 2022-03-16:
Confirmed and issue also affects 'Act' and '!Act'

elecpower, 2022-03-16:
model.yml
>logicalSw:
  0:
    func: FUNC_AND
    def: "!RADIO_ACTIVITY,RADIO_ACTIVITY"
    delay: 0
    duration: 0
    andsw: "!RADIO_ACTIVITY"
  1:
    func: FUNC_OR
    def: TELEMETRY_STREAMING,!TELEMETRY_STREAMING
    delay: 0
    duration: 0
    andsw: TELEMETRY_STREAMING

[image]

PR #1708 (2022-03-20): Companion fix reading of Act and Telemetry switches
Fixes #1697
closed 2022-03-20 by pfeerick