EdgeTX/edgetx #1700: EdgeTX 2.6 Cpn + Zorro: message 'File board ... does not match current profile board

VaBarefootBoy, 2022-03-17:
Issue previously reported by
[RC-SOAR](https://github.com/RC-SOAR) commented [on Feb 1](https://github.com/EdgeTX/edgetx/issues/1541#issue-1121278357)
reported 
Fixed by [#1572](https://github.com/EdgeTX/edgetx/pull/1572)

When transferring models from Zorro transmitter into ETX Companion 2.6 I get an error message, also regarding “File board (Radiomaster TX16S) does not match…”
To reproduce:
	1.	Connect to Companion via USB
	2.	Menu 'Read Models and Settings from Radio'
	3.	Then I get this message
	
[image]

At first I selected NO and got 

[image]

I went in an confirmed I had setup a NEW radio for the ZOR

- - - Radio Type Radiomaster ZORRO

I tried again selected “YES” and it seemed to work

BUT  after on session while changing mixes in a quad…..

When I disconnected and went into BetaFlight I found my 
Rudder channel had become pegged to the RIGHT.

I looked at the ZORRO Ch map and indeed my Rudder input was completely to the right ,
pegged at +100
and any stick movement was disregarded or showed no monitored movement.

I tried another Model but the same Rudder Input was there as well.
I tried Calibration but can not Center Rudder.
I tried re-flashing with no positive results.
I deleted Ch1 RUD and the re-program ch as RUD.
The TRIM will move the down as FAR AS +74 - - BUT
STILL NO MOVEMENT WHEN THE STICK IS MOVED.

Presently I am left with an inoperable ZORRO Rudder Ch.

All firmware and SD were backed up as they came shipped if any of that might come in handy.
 
Thanks for you help in advance.

eshifri, 2022-03-17:
Fix #1572 went into 2.6.1 nad, consequently, into 2.7.
2.6 has the bug.

VaBarefootBoy, 2022-03-17:
Maybe not you territory but do you have any suggestions on how to get the RUD ch, Ch 1 on this ZORRO FIXED / unscrewed ??

On the FliPhone
Godspeed & God bless
Stephen B. Caldwell

eshifri, 2022-03-17:
To be honest, I'm not sure.
You can try to use nightly build.
Or fix the model on the TX without companion.
O have never had Zorro in my hands, so ....

raphaelcoeffic, 2022-03-17:
@VaBarefootBoy Just in case: have you tried calibrating the sticks? Does it move during calibration?

VaBarefootBoy, 2022-03-17:
Thanks. 
I am going to give a nightly build a try but I think something has been corrupted in the ZORRO TX. 

I will as around. 

On the FliPhone
Godspeed & God bless
Stephen B. Caldwell

VaBarefootBoy, 2022-03-17:
> @VaBarefootBoy Just in case: have you tried calibrating the sticks? Does it move during calibration?

Yes I went to Calibration. Ch1, Rud, the monitor shows he stick left stick/RUD full right +100. the sticks can not be centered to begin calibration - well at least they can not be shown centered. 

I will try it with the monitor still far right and see what that does.

eshifri, 2022-03-17:
Are we talking about channel monitor or HW calibration?
If the stick cannot be centered in the HW Calibration window, most likely, it is a HW problem. (Maybe a cable connection.)

closed 2022-03-17 by VaBarefootBoy

pfeerick, 2022-03-17:
Try this build of 2.7 companion for Mac - it's the latest build at present -  https://github.com/EdgeTX/edgetx/actions/runs/1997574086

VaBarefootBoy, 2022-03-18:
I might give it a go. Thank you. 

I re-flashed the Zorro w the latest 2.6, & LUA and replaced my 1.3.9 Companion and after all that they began to work to get her. 

The official 2.7 comes out at the end of the month, I think, so I might just hold off for it to be released. 

Y’all have been very helpful. 
Thank you very much. 

On the FliPhone
Godspeed & God bless
Stephen B. Caldwell

pfeerick, 2022-03-18:
Yes, it was scheduled for 20/MAR but will most likely be thinking of first release candidate next week... glad it is all starting to work properly now though :)

brunweil, 2022-11-20:
Slightly different message:
"settings files board (RADIOMASTER TX16S) does not match current profile board (XXX)"
XXX = What I put in the radio setup model, connecting a RADIOMASTER TX12 MK2 ELRS.
The same occur on Mac OS 10.15 and Windows 10 with EdgeTX companion 2.7.1

pfeerick, 2022-11-21:
This is because the TX12 MK2 is not supported until 2.8 - you need to be running a 2.8 Companion (and firmware). Due to legacy reasons when there is a mismatch Compaion always complains about it being TX16S, rather than the value in the actual settings file.