EdgeTX/edgetx #1701: Radiomaster Zorro switch options

breadoven, 2022-03-17:
Couple of suggestions to improve controls on the Zorro:

1) Add option to change direction of menu scroll?/roller wheel
2) Make Telemetry button a toggle to switch telemetry on and off (rather than having to use the return button)

pfeerick, 2022-03-17:
1) is in progress - will probably be merged in the next couple of days. https://github.com/EdgeTX/edgetx/pull/1690
2) Sounds like a good idea... I don't think the TELE button has any secondary functions, so that sounds like a great idea! :)

breadoven, 2022-03-26:
So I tested the invert rotary encoder in 2.7 RC1 ... and it does just invert everything including the horizontal movement. So now the vertical movement is fine for me but the left scroll goes right which is more confusing than the issue with the vertical movement. So this is a no no for me. Really needs an option to invert just the vertical.

I did notice the Zorro just does the same as the other radios, left scroll goes left and up (invert OFF). Is this suppose to mimic navigation when reading text, English at least ? Perhaps this is OK when the axis of the rotary encoder is vertical, as it seems to be on most of the radios, but on the Zorro the axis is angled so left is also actually slightly down. Maybe this is why I keep trying to go the wrong way for vertical scrolling.

As for the Tele button. I'm thinking ON first press then cycle through screens with each additional press. Long press to turn OFF. No need to use any other buttons.

pfeerick, 2022-03-26:
> I did notice the Zorro just does the same as the other radios, left scroll goes left and up (invert OFF). Is this suppose to mimic navigation when reading text, English at least ? Perhaps this is OK [...]

It probably is that. I've been using the TX16, and X9D+2019 for so long that scrolling left to right for down, and right to left is up, so the ... say 30 degree tilt of the Zorro doesn't distract me want wanting left to right to still be down ,and right to left to still be up, although I can see the argument for it being the reverse due to physical orientation. The TPro is somewhat the same in that regard. As I said before, given the current code I don't know if if vertical scrolling can easily be separated from all the other input code, but I will look at the possibility in the future.

> As for the Tele button. I'm thinking ON first press then cycle through screens with each additional press. Long press to turn OFF. No need to use any other buttons.

For all other radios, and the Zorro is included in this, there are dedicated page keys which move from telem screen to telem screen... as well as control direction of movement through the screens.

ducatiJake, 2024-01-24:
Hi Guys, I was having issues with the rotary encoder seeming to move the UI in the wrong direction in some instances with my Zorro.  I stumbled on the I think new setting (V-I H-N) for rotary encoder which seems to always make the UI move in the direction I expect.  Not sure if this add was part of this thread, but it works for me on my new Zorro.  Now, related to this I am having problems with the perceived directions of the SB, SC, SE, SF switches. I always use the left rear switch for my gear, that's SE on the Zorro. As the Zorro is now, looking down from the top, pushing SE down toward the front of the radio is indeed interpreted by the SW as switch down. But, when using the radio, pulling SE up toward the screen which I would expect to do to get my gear up is again interpreted as switch down. Just the orientation of the switches on the radio.  I have the same issue with SB which I always use as my flaps.  What feels like commanding flaps up is interpreted by the SW as switch down.  Same is of course true for SC and SF on the other side of the radio. It would be great to add another option to the Hardware tab button selection options which are now. None, Toggle, 2POS, and 3POS. Add 2POS-REV and 3POS-REV which would just reverse the sense of UP and DOWN for those switches.  This would make my Zorro much more intuitive and prevent me from opening it up and rewiring the switches. I could of course just change the programming in my Zorro to deal w these differences, but I use the Zorro with the same models I use my TX16S with and thought I could copy model definitions from one to the other relatively seamlessly,  But this issue complicates that a great deal.  Thanks for your consideration.  John

pfeerick, 2024-01-25:
> Now, related to this I am having problems with the perceived directions of the SB, SC, SE, SF switches. I always use the left rear switch for my gear, that's SE on the Zorro. As the Zorro is now, [...]

You should always be considering the switch direction as as up or down, rotating the radio as necessary so the switches are on the right plane. i.e. if there were front switches, they would naturally be up or down. Top switches like on the zorro require you to rotate the radio so that the gimbals are facing the ground... then the switches are again correctly up and down (or away [up] or towards you [down[). 

2.10 is bringing some inversion options for inputs (pots and sliders) and FLEX switches (when available), but this is another argument for this to be made available for all inputs generally. If you're using input lines with the switches (which I would recommend when copying between different transmitter types), you can flip them by inverting the weight, but that wouldn't help with any mix lines using the switches directly.