EdgeTX/edgetx #1718: info line shows only 3 instead of 4 widget

TheIsotopes, 2022-03-21:
**Describe the bug**
since #1089 the info line shows only 3 widget instead of 4 as before, on my tx16s

**To Reproduce**
using latest master 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX 2.7.0 latest master
 - RM TX16S

pfeerick, 2022-03-22:
This is mainly due to the GPS icon also being enabled dynamically at runtime now, which is to the left of the battery and speaker icons... but I'll see if it's possible to get that space back if the GPS mode is not enabled.

TheIsotopes, 2022-03-22:
aaahhh, ok. so not a bug. Thanks for clarification.

closed.

closed 2022-03-22 by TheIsotopes

pfeerick, 2022-03-22:
Well, lets just say unintended side effect... :grin: ... easily fixed... you have can the forth zone back if the internal GPS is not enabled, and it will be taken away again if it is as that space is needed. Some better management (and options?) will be needed if remaining unimplemented icons are actually implemented... i.e. receiver battery icon, model altitude

commit 08ffcc72 (2022-03-22): fix: Only remove topbar widget zone if GPS active
Fixes #1718