EdgeTX/edgetx #1722: MPM Config protocol cannot be selected

TheIsotopes, 2022-03-22:
**Describe the bug**
MPM Config protocol cannot be selected on TX16S. 
If set in Companion, only `---` will be displayed but it will not appear in the list.
Should actually be at the top of the list.

This does not occur on my Zorro, it is displayed and can also be selected.

**To Reproduce**
using latest master

**Screenshots**
[image]

**General information**
 - EdgeTX 2.7.0 from latest master
 - RM TX16S

pfeerick, 2022-03-22:
Is it actually valid to be selecting config mode via the UI? I thought it was only for internal use? Otherwise, that can be easily fixed... (alternative was that at present a garbage value was being shown in the menu on colorlcd).

TheIsotopes, 2022-03-22:
> Is it actually valid to be selecting config mode via the UI?

It should already be selectable via the UI. This is required for the`Multiconfig.lua` script.

> And it won't appear at the top of the list if it is re-enabled, but under 'C' since the list is sorted alphabetically

Alphabetical sorting doesn't really matter, in the Zorro it's at the top of the list and I think it was the same with the TX16S with OpenTX.

pfeerick, 2022-03-22:
> This is required for the`Multiconfig.lua` script.

Actually, no, as of the [last update to the Multiconfig lua around nine months ago](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/tree/master/Lua_scripts), the lua script enables config mode directly itself, and then reverts to the configured protocol when exiting the script. Which is probably why the MPM doesn't return Config as supported protocol when queried for the list of protocols, and is also conveniently missing in the [protocols list](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Multi.txt) (you'll notice 86 is skipped over). 

So I can either add it (is there any other valid reason though?), or remove it from companion as it shouldn't be visible to the end user.

TheIsotopes, 2022-03-22:
I use my own model for this.
if I start the `MultiConfig.lua` script in another model where e.g. frsky is activated, the protocol was changes but the script returns the error `No Config telemetry...`

pfeerick, 2022-03-22:
Fair enough, I'll see how to go about adding it in then... in one place it's easy (when scan is invalid - but that probably won't help here)... not sure about as part of the dynamic protocol scanning, haven't looked at that part of the code yet. Problem for tomorrow ;)

TheIsotopes, 2022-03-23:
I took a closer look and researched it again, and I found another solution that works without a created model. Furthermore, I have also come to the conclusion that this should not be included in the protocol list, as you have already suggested.

pfeerick, 2022-03-24:
Would you mind elaborating on what this other solution is, just so it's here if anyone else stumbles on this issue?

Until the MPM Lua is updated to have a proper exit routine, which resets the MPM back to it's initial protocol, I'm now actually thinking it might be a good idea to still have the Config protocol/mode selectable, as at the moment the radio is being left in config mode when leaving the lua, thus the GUI is a bit borked...

TheIsotopes, 2022-03-24:
Sorry, I was too hasty. Unfortunately, my solution was not successful.
After long and many tests I found out that changing the protocol in `MultiConfig.lua` doesn't work very well. Every time the error `No config telemetry...` occurs, the protocol was changed but the status showed `Prot. invalid` (should show the firmware version). Then the script is aborted by this issue and the error `No config telemetry...` is displayed.

The question now is who is the trigger, is it EdgeTX/OpenTX, the MPM firmware or the `Multiconfig.lua` script)? 
Until a solution is found, as they say, let's add it to the list.

pfeerick, 2022-03-24:
I'm not 100% sure yet either... as long as the internal module is enabled (or external, if using external MPM) on TX16S, for me going into MultiConfig.lua works just fine... without it needing to be in Config mode... (i.e. I had it in Flysky 2A and Frsky modes) and **short press** of return will allow the script return to prior set protocol, and exit. A long press as the normal force exit will instead leave config enabled... I had to look at the code and find the Config_Release behaviour before I realised it needed a short press... intuitive IMO... better to have a "Exit" menu item to encourage using that. But that is a different matter.

TheIsotopes, 2022-03-24:
I did further research and found that protocols without a sub-protocol works without any problems.
For protocols with multiple sub-protocols the first/default sub-protocol works without problems.
As soon as i select another sub-protocol the error `No config telemetry...` comes up, if i go back to the first/default sub-protocol, everything works normally again.

pfeerick, 2022-03-25:
I can reproduce this on both EdgeTX AND OpenTX... i.e. Hitec Optima (MultiConfig works) vs Hitec Minima (MultiConfig: No Telemetry)... so it looks like MultiConfig lua also needs to preseve and then reset the subprotocol as well.

TheIsotopes, 2022-03-25:
I have tested this on TX16S and Zorro (both running EdgeTX nightly) and can confirm that everything works as expected.

Thx @pfeerick Now you can remember to remove the config protocol from the selection list in the radio and companion.

PR #1810 (2022-04-14): Companion remove multi protocol Config from list
Fixes #1722
closed 2022-04-14 by pfeerick