EdgeTX/edgetx #1723: Throttle to have a separate trim step

craigywsm, 2022-03-22:
When flying jets I want an option so that the trim step on the throttle can be adjusted 
e.g. you can go to min trim and full trim with 4 clicks of the trim adjuster.
This allow for easier starting of turbines and shut off of turbines

JimB40, 2022-03-26:
@pfeerick @raphaelcoeffic 
This is the case we talked about on last PSC meet.
Trims were coded mainly and solely for fine Aileron,Rudder or Elevator or Throttle adjustments.

 While there are other use cases when those inputs should br rather treated as multi-position swich.
 My intitial concepr was that you can set for particular model (or generally for radio in Hardware tab) that certain trim button will work as 2,3,4,5 or 6 position switch.  
Once lets say 5-posistion switch option is selected presing trim +/- will change output between 1000,1250,1500,1750 and 2000 us.
If 2-position mode is selected there will be only two states 1000us and 2000us
If 3 pos 1000,1500,2000
etc

This would allow to convert trims into multi function buttons.

gagarinlg, 2022-03-26:
That's something for the future expert mode ui. Maybe we should start discussing a future user interface, with a proper UI planning.

craigywsm, 2022-03-26:
thanks for the input.
here is a video on how ethos  has seperate trims on each of the controls. 
https://youtu.be/318Q0zSPN7U

My old graupner did something very similar. 
I would be more than happy to donate to the cost of who could develop this.

craigywsm, 2022-03-26:
also as I work in software support myself more than happy to help with planning and testing

pfeerick, 2022-03-27:
> While there are other use cases when those inputs should br rather treated as multi-position swich.
> My intitial concepr was that you can set for particular model (or generally for radio in Hardware [...]

Perhaps instead of that, we should look at what already exists... curves ;) If the trims were made separate, then the current "trim step" option could also have a "custom" or "curve" option, and then a curve applied to them, which would provide all of that 2-3-4-whatever steps. As to whether it should be global as well as model level, since it can now be saved to a template, I think model level is sufficient if it is needed for multiple models, as well as not cluttering up the radio setup UI and having to do double maintenance. 

At the present time, what you could do is disable the throttle trim via flight mode 0, setup a GV with your desired trim movement range, have two SFs with GV adjust + on Tu, and GV adjust - on Td, and then add that GV as an offset to your throttle. It's not as pretty / simple, but it should do the trick.

JimB40, 2022-03-27:
Peter sorry but you mind is bound to plane user-case :)

Of course you can use curves or make any other workaround. But this is like using screwdriver to "hammer" a nail.
It's possible but let's say very unconvinient in everyday use.

Don't have analytics tools but I bet more than 50% ETX (OTX maybe too) user base is drone pilots nowadays.

So... quad pilot don't use at all trim buttons in a way they were hard coded at all.
Instead of this, we switch off trims not to mess 4 basic channels plus lots of us use Logic Switches, Inputs with MAX and GV to change those unused Trim Buttons to simulate behavoiur as  ... Buttons. :)

I understand this is leagcy but having in mind that during last years share of type of users changed, IMHO it's highest time to consider to rework those T1,T2,T3,T4(T5,T6) to work as Buttons :) then to give user possibility to define how it behaves. 
One way of usage , will we fine trimming of four standard channels.

pfeerick, 2022-03-27:
And why would it be plane based, when I predominantly fly quads? :-P

I think you're missing the point, which is who are you to tell me that I can only have these abritary 2pos, 3pos, 4pos positions with fixed positions. Here I could acuse YOU of being too drone focused :grin: ... and not seeing the possibility of curve points being used to set the number of positions,  what those positions are, and what order they are in.  Which AFAIK is not currently possible (or if it is, it would require some fancy lua or LS/SF/GV magic to make it work). 

But I can see how that could be an even more advanced option as well.

jmxp69, 2022-06-20:
Probably not as succinct as a firmware option to re-configure, but it does allow the trim switch to be used a 4pos or xpos switch.
https://youtu.be/NvIKHa90x2k