EdgeTX/edgetx #1738: Build for NV14 fails

eshifri, 2022-03-26:
**Describe the bug**
Build for NV14 fails for both firmware and simulator.
The error message is:
 Building CXX object radio/src/CMakeFiles/firmware.dir/pulses/afhds2.cpp.obj
D:/msys64/home/eshif/edgetx/edgetx_main/radio/src/pulses/pulses.cpp: In function 'void extmoduleSendNextFrame()':
D:/msys64/home/eshif/edgetx/edgetx_main/radio/src/pulses/pulses.cpp:705:7: error: 'extmoduleSendBuffer' was not declared in this scope
  705 |       extmoduleSendBuffer(extmodulePulsesData.afhds3.getData(),

**To Reproduce**
Run build for NV14

**General information**
 - EdgeTX current version)
 - Windows MSYS2 env.

pfeerick, 2022-03-27:
I can't reproduce this... try clearing your CMakeCache.txt, and re-running cmake... I used 
```
cmake -G "MSYS Makefiles" -Wno-dev -DPCB=NV14 -DLUA=Y -DDEFAULT_MODE=2 -DUNEXPECTED_SHUTDOWN=N -DDEBUG=N -DCMAKE_BUILD_TYPE=Release ../
```
and 
```
cmake -G "MSYS Makefiles" -Wno-dev -DPCB=NV14 ../
```

without issue on current main. Since pulses is playing up and was recently edited, something is probably not up to date. 

[image]

[image]

eshifri, 2022-03-27:
Yes, cleaning cache fixed it.
Thank you!

closed 2022-03-27 by eshifri

pfeerick, 2022-03-27:
Fantastic! Yeah, cmake and the cmake cache can be a real @#*& sometimes...

Ktr128, 2022-03-27:
Faced the same problem: 
(/workspace/edgetx/radio/src/pulses/pulses.cpp:705:7: error: 'extmoduleSendBuffer' was not declared in this scope 705 | extmoduleSendBuffer(extmodulePulsesData.afhds3.getData(), )

Thanks for the help!