EdgeTX/edgetx #1740: Re-code behaviour rotary encoder (roller) input

JimB40, 2022-03-27:
This is specyfic to radio models with rotary encoder placed at angle (Zorro, T-Pro, Tango)

There are different cases when direction should be implemented diferently too feel mor naturally

1. Horizontal UI change (Right -> Right, Left -> Left)
2. Vertical UI change (Right -> Up, Left -> Up)
3. Value change (Right -> Higher value, Left -> Lower value)

Related disscussion:
#1690 #1701

breadoven, 2022-03-28:
I managed to get this working on a Zorro as Left scroll = left and down.

Also out of interest changed the vertical scroll behaviour so it doesn't move along a row with multiple settings, e.g. Radio time/date setting, when moving down rows using Left scroll. It only moves along the settings in the row when you Right scroll. So you have to move down using Left scroll to the multi setting row you want to change and then use Right scroll to select settings in the row (Left scroll will move back along the row but only if entered initially using Right scroll). I think I prefer this since you always start at the top of a new page and most probably just want to move straight down to the row you want without having to cycle though all the settings above. Works well for the inverted vertical scroll behaviour but not so well for the current scrolling behaviour (Left = left and up) because you could only skip row settings when moving up rows rather than down losing any benefit for a new page starting at the top row.

JimB40, 2022-05-13:
To be closed @pfeerick ?

pfeerick, 2022-05-13:
No. Do *not* close issues before the PR is merged! 🤦

pfeerick, 2022-09-09:
Was re-done and implemented, for B&W *only* at present, in #2045

closed 2022-09-09 by pfeerick