EdgeTX/edgetx #1756: NV14 Throttle and Rudder issue

Outwide, 2022-03-29:
Using mode 2, throttle and Rudder issue. I found in the Channel Monitor the appearance that the throttle and rudder seem to stick or was stiff just a bit compared to the Elevator and Aileron which was very smooth and appeared to have no problem, looking at the graphics representation and with stick control. Reference report #1630. 
I then found that the throttle set to at about ¼ throttle speed movement the Rudder stick left or right and you got interference with the Throttle of the motor slowing down. I thought it may be Beta flight or Inav, software related, so I changed from flight controllers to the standard Receiver FS-iA6B. 

To Reproduce; I perform this test 6 times plus to reproduce the error. I then tried with the throttle position higher and was able to hear a difference in tone change when moving the Rudder stick left or right and the motor slowing down. Note; that when you bring the Rudder stick back to centre the motor returns back to its correct speed. Also it may be related to reference #1605 
I am using a T-motor and T-motor ESC matched. 

Expected behavior: Moving the Rudder stick left or right should not interfere with the Throttle position or performance of the motor (basic standard setup)

I like Edgetx.

General information
EdgeTX "Black Pearl" v2.7.0-RC1
FLYSKY NV14 
FS-iA6B

Outwide, 2022-03-29:
In relation to the above I have just connected my second FLYSKY NV14 with Opentx 2.18 and found that the Binding is very easy and quick to FS-iA6B. In Edgetx 2.X it has taken anywhere up to 4 goes to Bind using the same procedure.
On the FLYSKY NV14 with Opentx 2.18 with the throttle and Rudder issue I found that at about ¼ throttle speed movement the Rudder stick left the motor would slow down, “a lot” or moving the stick to the right the motor would speed up a little bit.

Also battery usage on the OpenTX appears to be lower even if I swap the batteries and recharge and test again.

I like Edgetx.

General information
EdgeTX "Black Pearl" v2.7.0-RC1
FLYSKY NV14 
FS-iA6B

pfeerick, 2022-03-29:
Just to be clear, you're certain there is no physical stick movement happening? I have noticed that the thottle/rudder gimbal does seem a little out on the NV14... IIRC, moving from high to low throttle the rudder seems to wander from one side to the other.

For the binding, is this when putting the tx in bind mode first, then powering up the receiver in with the bind plug in? As not doing it in this order can be rather flakey. Receiver will usually nearly instantly connect/bind.

Outwide, 2022-03-29:
Hi Peter, using Edgetx 2.7 I was very careful and did this a number of times and I am confident NO stick movement in the Throttle direction.  I am happy to bring the NV14 systems an test bench up to you  so you can see, I am in Brisbane.
I think you have read the other information on the Opentx as well.

For the binding, the receiver is set to Binding mode First and then I activate the transmitter Second to bind and it connects, Opentx very quickly beeps once done and Edgetx not so quickly and may or may not beep, sound like it is trying to connect but not, unusual.

Note; I have 2 NV14 transmitters and check between both.

I like Edgetx

Ktr128, 2022-03-30:
This is due to the absence of the "Dead zone" setting in EdgeTX.
This setting in OpenTX for NV14 allows the user to set a dead zone in the center of the sticks.

Outwide, 2022-03-30:
Hi I ask for information on the dead zone for opentx, how was this setup as I did not know of it. Is there a YouTube video or some notes. Please advise.

Ktr128, 2022-03-30:
This setting is in:
Radio settings -> hardware
The default value is 4, but it can be changed by the user.

raphaelcoeffic, 2022-03-30:
I would need to take a look at how this is implemented in the FlySky fork, but I believe this should be quite simple.

Ktr128, 2022-03-30:
From mixer.cpp:
#if defined(STICK_DEAD_ZONE)
if(g_eeGeneral.stickDeadZone && ch != THR_STICK){
if(v > P_OFFSET)
{
//y=ax+b
v = (int)((aParam * (float)v) - bParam);
}
else if((v <= P_OFFSET) && (v >= N_OFFSET))
{
v = 0;
}
else if(v < N_OFFSET)
{
//y=ax+b
v = (int)((aParam * (float)v) + bParam);
}
}
#endif

Outwide, 2022-03-31:
Hi thankyou, please do. 👍

PR #1777 (2022-04-06): NV14: Deadzone option
Summary of changes:
- adds deadzone option from open-flysky/Flysky-OpenTX-Test-Branch to radio firmware
- teaches Companion about it enough to not nuke the radio setting

~~Currently, untested.~~ Is now upgraded to status of "deadzone works, but monkey implementing settings storage screwed up". Weekend problem. 

Also, radio hardware option would probably be better as a `Choice` rather than `NumberEdit` since it is limited to 7 options. However, it should be sufficient to see if this is the resolution. 

May resolve #1756
closed 2022-04-06 by pfeerick