EdgeTX/edgetx #1764: Cannot flash external modules on Tlite with 2.7.0 RC1

phileville, 2022-03-30:
**Describe the bug**
I am not able to flash my external LRS module on the Jumper T-lite after updating to the 2.7 RC1.  When attempting it gets stuck on reseeting device and then displays "Device not responding" and the fails.  When I rollback to 2.6.0, I am able to update the external module as expected.

**To Reproduce**
Update radio to 2.7 rc1
Attempt to flash external module firmware from the file explorer on the radio.
It will fail

**Expected behavior**
It should complete flashing as normal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version 2.7.0 RC1
 - jumper T-Lite, Ghost nano module

pfeerick, 2022-03-30:
So this is a Ghost Nano module, which you are running ExpressLRS on? And for updates it's flashed via the transmitter (like the HappyModel_ES915TX)? 

Unfortunately I won't be able to reproduce myself as my T-Lite ELRS is wifi or USB update only, so hopefully something will jump out in the code as the culprit.

phileville, 2022-03-30:
Yes, it is flashed with ExpressLRS and I have to flash it either with the transmitter or the ST link. Like I mentioned, it seems to hang when initially trying to reset to flash...then it gives the device not responding error and fails.

raphaelcoeffic, 2022-03-30:
I could probably try with my MPM-lite, it uses the same protocol. By the way, this is using software-serial on the reception side, so it could be a bit touchy.

raphaelcoeffic, 2022-03-30:
@phileville phil, forget about what I said: it is **not** using software serial, it is done on the S.PORT pin, which is far better as it is hardware serial rather than software serial.

pfeerick, 2022-03-31:
> **iRange iRX4-lite**

Damn you... when I was looking for Nano form factor MPMs I only saw the TBS one... now I have to go an order this one! 😆

JrminLin, 2022-04-09:
My X9D+ with Namimno FLASH can't flash ELRS FW via the transmitter on 2.7 RC1 RC2 too, but 2.6 is OK.

"Firmware update error
Device not responding
[EXIT]"

deadbytefpv, 2022-04-17:
Same on my TX16S with the R9M ACCESS flashed with ExpressLRS

Firmware update error
Device not responding

R9M would restart as indicated by the ELRS start-up tune, but then the message would appear.

a1j, 2022-04-23:
Confirmed, flashing is not working on 2.7. Downgrading to 2.6 fixes that.

Fumes74, 2022-04-26:
Same here, TX16S with R9M, can't upgrade on 2.7.

Firmware update error
Device not responding

Kaggen67, 2022-05-06:
Can confirm that flashing R9M 2019 Access module (RM TX16S) on 2.7.0 don't work, but works on 2.6.0. Unfortunatly i didn't see this issue before i posted my own. :/
https://github.com/EdgeTX/edgetx.github.io/issues/44

a1j, 2022-05-06:
There is no power enabled on the expansion interface during flashing.

PR #1944 (2022-05-07): fix: flashing ELRS modules would use the wrong port driver
Please note that `main` is not impacted, as the code has already been fixed because of something else.

Resolves #1764
commit 946eb880 (2022-05-07): fix: flashing ELRS modules would use the wrong port driver (#1944)

Resolves #1764
closed 2022-05-07 by pfeerick