EdgeTX/edgetx #1766: [Serial Refactor] 2.7RC1 does not power TX16S AUX uarts for all protocols

stronnag, 2022-03-30:
On a TX16S, the AUX UARTS are not powered for all protocols

| Protocol | Port Power |
| ----------- | --------------- |
| Telem Mirror | Yes |
| LUA | Yes |
| GPS | Yes |
| Telem In | Yes |
| CLI | No |
| Debug | No |

Most recently tested with 5ffe75729 2022-03-30

The AUX UARTS should be powered for all protocols (e.g. to use bluetooth dongle).

stronnag, 2022-03-30:
This has nothing to do with BT in the firmware, it's using an external BT dongle on a AUX port --- transparent serial. 
So if I want to use a BT dongle for debug or CLI when I don't have a VCP cable available, you're telling me I can't.

raphaelcoeffic, 2022-03-30:
@stronnag please check here:
https://github.com/EdgeTX/edgetx/blob/777b0a48ae5e453e09a2fc6d8e9b920119c77f76/radio/src/serial.cpp#L192-L250

And pay special attention to these lines:
```c++
power_required = true;
```

stronnag, 2022-03-30:
OK, so it is a feature that CLI and DEBUG are not available  for such devices.

pfeerick, 2022-03-30:
It's more that that is an unexpected use case. As Raphael said though, this is *not* set in stone. There is no reason a UI option could be added to allow for the power to be toggled on or off in those modes if there is enough interest for such an addition.

rotorman, 2022-03-30:
Indeed, I can relate to the original post - if the serial port is not OFF, power should be turned on, irrespective if the usage is DEBUG or CLI.

raphaelcoeffic, 2022-04-01:
@rotorman feel free to change it ;-)

rotorman, 2022-04-19:
@stronnag would you care to test #1848 

[image]

stronnag, 2022-04-19:
> @stronnag would you care to test #1848
> 
Sure, I'll get back to you soonish. Thanks.

stronnag, 2022-04-19:
On my TX16S
* AUX ports are powered by command in the UI for all protocols
* Don't seem to be able to set a working CLI on AUX1/2; other tested protocols (LUA, TelemMirror are OK)
* CLI works on VCP

stronnag, 2022-04-19:
> Thank you for testing!
> 
> > On my TX16S
> > ```
> > * AUX ports are powered by command in the UI for all protocols
> > ```
> 
> Meaning you can enable and disable power OK, as per checkbox?

Yes 

> > * Don't seem to be able to set a working CLI on AUX1/2; other tested protocols (LUA, TelemMirror are OK)
> > ```
> 
> This is likely unrelated to #1848 For TX16S need to use 400.000 baud 8N1, other radios use mostly 115.200 baud 8N1. Did you try with 400.000 baud?

No, my BT dongle is locked to 115200. I suspected that would be the problem. Let me try a USB-TTL adapter.

stronnag, 2022-04-19:
Took a while to find an USB-TTL dongle that allows non-traditional baud rates (i.e. accepting 400000 rather than deciding that 460800 is close enough). No difference though, still no CLI on AUX; so for a different issue (at least if it's the same on MacOS and Windows).

Not withstanding your efforts, perhaps we should just close this again and accept that CLI is not a option for AUX (and the UI should enforce this limitation to avoid user frustration)?

rotorman, 2022-04-19:
@stronnag Just tested #1848 with TX16S mkII, AUX1 set to CLI, [SparkFun DEV-09717](https://www.sparkfun.com/products/9717) hooked up to AUX1, `[hTerm](https://www.der-hammer.info/pages/terminal.html)` connected with 400.000 baud 8N1. Works beautifully for me.
Are you sending CR+LF at the end? Type `help` accompanied by CR+LF? This is the output I get:

```
> help
beep [<frequency>] [<duration>]
ls <directory>
read <filename>
readsd <start sector> <sectors count> <read buffer size (sectors)>
testsd 
play <filename>
reboot [wdt]
set <what> <value>
serialpassthrough <port type> <port number>
help [<command>]
gps <baudrate>|$<command>|trace
>
```

stronnag, 2022-04-19:
I have the CR+LF.
Works fine over the VCP, USB-TTL device works fine at 400000 with loopback. TX16 Mk1, Arch Linux (5.17), firmware compiled with gcc 11.20. I'll try an older compiler tomorrow.
However, this PR is fine, the power is working, before we too far off topic.

pfeerick, 2022-04-20:
Or take the compiler out of the equation - try the github build of the PR? I find going back to / sticking with the GH builds when testing removes environmental variables if there are potential issues. 😀 

https://github.com/EdgeTX/edgetx/suites/6172150719/artifacts/216510037
or 
https://buddy.edgetx.org/#/dev/flash/pr?version=pr-1848%405682286173f09de936ac0640cb4f90a48a70a66e

stronnag, 2022-04-20:
Thanks @pfeerick, the first option worked OK (CLI /AUX1 / 400000 baud). Interesting, ianv firmware has some serial issues with 11.20 as well.
For my own satisfaction, I'll have to give 10.20 a go.

stronnag, 2022-04-20:
and for the record, 10.20 works as well.
Thanks both.

closed 2022-04-24 by pfeerick