EdgeTX/edgetx #1767: Touch doesn't stop inactivity alarm

mha1, 2022-03-30:
Not really worth to mention ...

Radiomaster TX16s, fw 2.0.7-rc1
Inactivity alarm set 15 minutes

eshifri, 2022-03-30:
AFAIK inactivity alarm is looking at he controls (steaks and switches) only.

mha1, 2022-03-30:
Yes, but with touch there is one additional means of  input. My use case: sitting in front of the radio and editing stuff. It starts beeping as if the radio was abandoned. Really no big deal as it is easy to push a stick. If it's easy to add touch reset the inactivity timer I'd do it.

eshifri, 2022-03-30:
I'm not arguing with you suggestion to fix this. :-)
I was just saying that this is not new: the same happened in OTX with physical buttons.

mha1, 2022-03-30:
This resets the inactivity alarm timer at any touch event. Works in a private build. Any concerns?

[image]

eshifri, 2022-03-30:
Looks good to me. Just submit a PR.