EdgeTX/edgetx #1768: View text and model notes have line breaks at 60% of screen width on TX16s

mha1, 2022-03-30:
Radiomaster TX16s, FW 2.7.0-rc1

View text and model notes break [TESTHoTT.txt](https://github.com/EdgeTX/edgetx/files/8381208/TESTHoTT.txt) much to early.

[image]

raphaelcoeffic, 2022-03-30:
Nice test text :-D I think we should wait for LVGL port to complete, as the library has support for all this, and we don't need to care ourselves about it.

eshifri, 2022-03-30:
This   is a side effect of using TT fonts: we need to know the length of the string before displaying it and with variable width of a letter we have to estimate for the worst case.

pfeerick, 2022-07-19:
ty for spotting... yes, closed by lvgl and #2070

closed 2022-07-19 by pfeerick

mha1, 2022-07-19:
Thanks to both of you