EdgeTX/edgetx #1769: Companion telemetry almost useless in versions 2.6 and 2.7

dwheeld, 2022-03-30:
In ETX version 2.5, Companion simulated telemetry performed the same as it did in the last few versions of OTX. It would simulate most of the Frsky telemetry sensors, while non frsky telemetry was not simulated. Now, with version 2.6 and 2.7, not even Frsky sensors are being simulated. The only thing showing in the simulated telemetry is RSSI.

Please roll the Companion simulator telemetry back to 2.5.

dwheeld, 2022-03-30:
Obviously something has changed. The way I understand it,  companion simulator runs in a 'shell' that uses the transmitter firmware,, so anything that is changed in that firmware could have an affect on it. (I'm using TX16S transmitter and windows 10)

Are there any diff files available for changes that have been made for 2.6/2.7? I would be interested in seeing what was changed.

elecpower, 2022-03-30:
@raphaelcoeffic from memory nothing changed deliberately though we entered the world of YAML. There maybe a relationship with #1650.

dwheeld, 2022-03-30:
I'm sure nothing was done deliberately but there were quite a few changes made for yaml and other things so it may be difficult to find. In the mean time, I will go back to 2.5 because I use companion and its simulator often. Probably hundreds or possibly thousands of times over the years, and have used yaml 1 time so will not miss it.

pfeerick, 2022-04-01:
For me 2.7 seems to be behaving exactly the same as 2.5 ... there are only two sensors visible, and as soon as any of the other sensors have their values changed from zero, they appear (with discovery running, of course). 

[image]

dwheel, 2022-04-01:
Yes. If you rediscover sensors, it works fine. But you can't use any programming based on sensors that are discovered in companion simulator.

Try a model that you downloaded from the transmitter which had all of the sensord discovered. Do not discover sensors in companion simulator. This is what you will see.

[image]

Try the same thing with 2.5 and this is what you will see. 

[image]

pfeerick, 2022-04-02:
Ah, thanks... that extra bit of info should help in figuring out what changed ;) Probably something simple around the sensor id parsing. What is annoying though is we want to make it so discovery doesn't effect existing sensors... Probably to the point that you don't even *have* to "discover"... it would always be active and you just enable the sensors you're interested in. But that will be a future version.

dwheeld, 2022-04-21:
It seems to be working in the latest release.

pfeerick, 2022-04-21:
Unfortunately I'm not seeing this on re-testing... without rediscovery, telemetry simulator isn't simulating, for Frsky X4R-II + VFAS + cell monitor - with either manually entered values, or replaying a log. Telemetry itself is fine on transmitter. 

Interestingly, the "discovered" simulator sensor ID and instances are different to those that were discovered via the transmitter. i.e. RSSI from transmitter
[image]
becomes (by default)
[image]

Now, thinking I could possibly just set the ID to 0, and RSSI would appear where it should.. nada, it vanishes. But if you set it to 1, it appears as a sensor with a 0 instance id... so for starters, I think were have a off by one index issue here... but I don't know what is going on with the sensor ID being different. 

Fresh import from TX16S running v2.7.0 fw using v2.7.0 Companion, MPM v1.3.3.7, D8Cloned.

dwheeld, 2022-04-21:
I only tried it on one model that uses an x8r receiver. It worked fine. When I get back home tomorrow I will try it on several others and let you know the results.

dwheeld, 2022-04-21:
You are correct. Versions 2.6 and 2.7 are still not working as they should. But, I have now discovered that the problem goes a little further than that. Version 2.5 is also not working as it should with some model files. My testing today has revealed some very confusing issues.

The transmitter used for testing (Eachine TX16S) is on version 2.5. My tests were done as follows:

1. Copy an existing simple model file to a new slot in transmitter.
2. Bind the transmitter/receiver.
3. Delete existing telemetry and then Discover New telemetry.
4. Connect transmitter to Companion 2.5 and 'Read Models and Settings From Radio' by clicking the appropriate icon on left side of Companion.
5. Double click on the desired model.
6. Click Simulate bar at bottom of page.
7. Go to Model Settings and click on Telemetry Icon.
8. Insert values into Telemetry Simulator VFAS and Curr boxes.
9. Verify these values are being shown in the simulated telemetry.
10. Select File/Save as, and save model file.
11. Close 2.5 Companion and open 2.7 Companion.
12. Select File/Open and select the file saved above.
13. Repeat steps 5-9.

Every model that worked in 2.5 also worked in 2.7. All that would not work in 2.5 also would not work in 2.7.

I only have 2 receivers that are not buried in aircraft, so used these for the tests. Both are X8R on ACCST v2. Simulated telemetry worked fine for both in the tests.

I have a model that has been working in 2.5 and included it in the attached .otx file. It is named RXSR. Simulated telemetry for this model also works in the 2.7 Companion Simulator.

Another test was performed using the second X8R receiver, but this time I created a new model instead of copying the simple model. Simulated telemetry does NOT work correctly with this model in Companion 2.5 or 2.7. It would not simulate VFAS or Curr. Analyzing the differences between 'X8R 2'  and 'X8R 2 new model' may yield a good clue to solving the problem.

Addition:

I borrowed an X4R receiver from a friend. Using a copy of the simple model file, companion simulated telemetry works fine with this receiver in both 2.5 and 2.7. I then bound it to a new Created Model. Results were the same as with the X8R newly created model. It would not simulate VFAS or Curr in companion 2.5 or 2.7

The .otx file for these models is attached below.

[20220421-each-test8.zip](https://github.com/EdgeTX/edgetx/files/8534165/20220421-each-test8.zip)

dwheeld, 2022-04-21:
pfeerick

I loaded the latest 2.7 firmware on my Jumper transmitter. After reading a model created on it into Companion 2.7 and then selecting the "Disable Multi Sensor Handling" check box, I can simulate VFAS and Curr telemetry. I am in favor of closing this issue. Do you agree?

closed 2024-04-09 by pfeerick