EdgeTX/edgetx #1778: Inconsistent acoustic feedback fro touch events on top bar

mha1, 2022-03-31:
Radiomaster TX16s, fw 2.7.0-rc1

Another minor one, maybe I haven't understood the acoustic feedback on touch event logic yet. Some examples:

1.  Main screen
touching the **left** half of the top bar (bringing up select model to model settings) provides **no acoustic feedback**, i.e. no beep on touch. touching again
touching the **right** half of the top bar (bringing upscreen setting to about edgetx)  provides **acoustic feedback**, i.e. beep on touch
touching on left or right half of top bar again (popup vanishes) provides **acoustic feedback**

2. In model select screen
touching model categories list or any model category provides **no acoustic feedback**
touching the leftmost symbol to return to main screen provides **acoustic feedback**

3. model settings screen
touching any symbol from model setup to telemetry provides **no acoustic feedback**
touching the leftmost symbol to return to main screen provides **acoustic feedback**

4. Widgets
touching a widget provides **no acoustic feedback**
touching a selected widget to show selection list provides **no acoustic feedback**
touching a selection list entry widget settings provides **acoustic feedback**
touching a color number field provides **no acoustic feedback**

pfeerick, 2022-03-31:
> touching the right half of the top bar (bringing up screens settings to about edgetx) provides acoustic feedback, i.e. beep on touch

This is a bug, due to the sneaky way showing that end of the menu was done... it opens then menu, and scrolls one to the 
left, which wraps around. Thus you get the beep you would get when pressing on the left side at the top, and scrolling one click left. 

> touching on left or right half of top bar again (popup vanishes) provides acoustic feedback

More than that, if you touch *anywhere* but one of the carousel icons, it'll beep and close. But same as the following...

> touching a selection list entry widget settings provides acoustic feedback
> touching the leftmost symbol to return to main screen provides acoustic feedback

Those I'm not sure of off-hand. Most likely something to do with focus and particular events. So you've decided to join a larger group of people, who have their Sound mode set to "All"? :grin:

mha1, 2022-03-31:
>  So you've decided to join a larger group of people, who have their Sound mode set to "All"?

Sure, "All" is my preferred mode, especially for my pizza pies :-)

mha1, 2022-06-27:
@pfeerick - hey yo 2.8 still beeps or not on touch with (to me) no logical pattern. What's the plan for consistent audio feedback for touch. If you ask me I'd like a click if my thick fingers are recognized touching the screen. This is how I know the touch was successful and a maybe delayed response is not due to a missed touch. This keeps me from immediately touching again if there is no click and no immediate reaction. (Alternative: guarantee a reaction time  less < 200ms :-))

mha1, 2022-07-31:
@pfeerick  - I checked a recent Companion build. There is a beep for opening lists but no beeps for touching buttons or icons. Work in progress or feature?

pfeerick, 2022-08-01:
Most likely just coincidence if there is any change - I'm not aware of any specific commits / work on this.

mha1, 2022-08-01:
@pfeerick is it planned to support accoustic touch feedback (beep or click)?

pfeerick, 2022-08-01:
I don't see why not... as you pointed out in this issue, it's inconsistent, and we can't have that ;) I suppose it needs some sort of tweak so that when a press is detected, it should beep/vibrate (depending on sound/haptic settings).

mha1, 2022-08-04:
@pfeerick Hi Peter, some time ago I implemented a PR to reset the inactivity alarm on touch events. It got implemented in 2.7.1. With LVGL this was rewritten using the same logic. I think this logic could be extended to not only check if the inactivity alarm counter needs to be reset but also to generate acoustic or haptic feedback if a touch event was detected. 

It would require a new function that checks if the radio setting allow audio or haptic feedback and if so issues a brief beep or buzz. If you think this is a reasonable and simple enough way to do it I'd need some help on which radio settings states to check and how to issue the feedback (e.g. audioQueue.playTone()?).

LvglWrapper.cpp/touchDriverRead() line 212:

```
// Reset inactivity counters
if (st.event != TE_NONE) 
  { reset_inactivity(); 
    touch_feedback();    // new function to beep or click depending on radio settings
  }

void touch_feedback() {
  // check radio settings
  // generate feedback
}
```

closed 2022-09-14 by mha1