EdgeTX/edgetx #1788: trim labels

djabmal, 2022-04-02:
i have a radiomaster TX16S  with edge tx 2.6 installed .In flight mode the trims are not labelled anymore (as in opentx) making difficult to know which trim we are modifying

djabmal, 2022-04-17:
hello do you think its possible to fix this little problem ?

pfeerick, 2022-04-18:
Yes, of course... it is just a matter of when and how. There is a major re-write planned for the UI over the course the next few versions of EdgeTX, so it will be certainly be addressed as part of that. In the meantime, I guess it *may* be possible to sneak in some text to the left of "Trims" that indicates which trim choice is in focus/active... I'll try and look at that next week.

djabmal, 2022-04-18:
Thanks a lot