EdgeTX/edgetx #1789: External Module unstable mixer period detection

PSi86, 2022-04-02:
**Describe the bug**
The CRSF status field (introduced in the latest nightlies) shows Telemetry errors in combination with a changing mixer rate.
While the radio sits on the desk every 1-2 minutes the errorcounter starts incrementing, then the Mixer Period goes to 500Hz, while the errorcounter keeps incrementing, then after an undefined ammount of time (3-10 seconds) the mixer period goes back to 250Hz and the errorcounter stops incrementing.
The behaviour is happening with all baudrates but with higher rates the errorcounter increments faster (during the error) than with lower rates. Communication to the elrs module via LUA works fine and the packet indicator there shows no errors.

**To Reproduce**
Set ELRS Module to 250Hz packet rate. CRSF status field shows "250Hz 1err".
Observe the CRSF status indicator.

**Expected behavior**
1. After setting a packet rate the mixer rate should remain constant during operation.
2. The recovery from the errorstate (continous increments on the error counter) should not take several seconds.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX 2.7 RC1 / nightly (8b515ed)
 - FrSky X9D+ SE 2019
 - Happymodel Pro Series 2,4GHz ELRS

PSi86, 2022-04-05:
Does anyone see a similar behavior? Of course it could be an issue with my handset, but the way it happens even on 400k and without having issues on the rx side I think it could be something in the software.

pfeerick, 2022-04-06:
I'm afraid not... 😢 

On TX16S /w 2.7.0-RC1 (289b11687807506c47c9ee3adbf8f4070845cb36), with ES24TX ELRS module, EP2 receiver, at 400k, 250 hz, after five minutes it still only shows 1Err. 

On X9D+2019 /w 2.7.0-RC1 (289b11687807506c47c9ee3adbf8f4070845cb36), same TX and RX, 400k /  250hz, after five minutes by the session time stats, still only 1Err. 
With OneBit on, at 912k / 500hz, after five minutes, only 1 Err.

However, I have noticed a different issue in my case... if the baud rate is changed the link to the module doesn't seem to be recovered... either need to turn the module/transmitter off and on again, or pull the module - but that could be the module not wanting to change speed after initial connection. At that point, it jumps to 2 Err and is still stable... :-/

ELRS 2.3.0 on TX, probably the same on the RX.

raphaelcoeffic, 2022-04-06:
@pfeerick the issue with changing baudrates and module not coming back is because this module needs a much longer power reset. The power reset is currently 200ms when changing baudrates.

pfeerick, 2022-04-06:
Oh, fun... so the same as the CRSF model change reset time issue? Ouch!

gagarinlg, 2022-04-06:
Maybe a new option to set the power off time when changing setting?
Probably just two values, normal and long, where long is 1s

PSi86, 2022-04-06:
Did more testing. At 400kBaud and Onebit enabled there are no errors. But even at 921kBaud with Onebit there are errors. Do I need to do the inverter mod? Thanks for your replies!

raphaelcoeffic, 2022-04-06:
Is the X9D+2019 borked just like the old one? Is that really possible?

PSi86, 2022-04-06:
I was running 1.87MBaud for several month now and never had an issue controlling my models nor did I have any bad packets reported in the ELRS Lua. Because of this it's hard to believe that there are so many errors in the communication to the external module and that they always only come in groups. It more looks like the ELRS Module receives the RC data from the handset but the handset does see errors while receiving data from the external module.

PSi86, 2022-04-07:
I would really like to try some different combinations of configurations to get a better feel for the behaviour of my hardware.
- Set GPIO Speed to high for all baudrates
https://github.com/EdgeTX/edgetx/blob/f475a9ce85f3f996bd7cc3a41afede9d54410956/radio/src/targets/taranis/telemetry_driver.cpp#L62
- Disable / enable USART Oversampling for all baud rates https://github.com/EdgeTX/edgetx/blob/f475a9ce85f3f996bd7cc3a41afede9d54410956/radio/src/targets/taranis/telemetry_driver.cpp#L69

What is the easiest way to compile a test build? Do you think it makes sense to test?

rotorman, 2022-04-07:
> What is the easiest way to compile a test build? Do you think it makes sense to test?

https://github.com/EdgeTX/edgetx/wiki/Building-radio-firmware-in-a-webbrowser-with-Gitpod

PSi86, 2022-04-07:
Thanks a lot @rotorman for the tip - didn't know that!
I did some test builds but changing the GPIO_Speed and oversampling8 option does not seem to make a difference at all.
Going to stick to 400kBaud for now but it still bugs me.

closed 2022-04-10 by PSi86

pfeerick, 2022-04-12:
Well, interestingly enough... I have my X9D+2019 (no hardware mods) set to 912k, OneBit is on, and at 500Hz after 8 minutes, it's up to 50 errors... however I don't think the link glitched at all... and about two minutes later I did get a telemetry lost followed by telemetry recovered, and 9-10 packets dropped in succession... jumping to 1.87M and module stopped talking after about 90 seconds... so it does look something is not right with the timing there. But I generally only go up to 250Hz anyway, and haven't had any issues before, so 400k/250Hz is where I'll stick for that hardware.

PSi86, 2022-04-22:
The traces to the external module bay are really long in case of the taranis. Not hard to believe that this limits the capabilities in terms of max bauds.. But I don't mind - important was to identify the issue in the first place and know at which bauds it works. Thanks again for adding the error counter!