EdgeTX/edgetx #1790: 2.7 RC1 - Global Function with action='no touch' fails

RC-SOAR, 2022-04-02:
Create a global function with Action = NoTouch and Switch = ON

_Expected_: touch is disabled 
_Actual_: touch remains enabled

If I create a special function with the same parameters, it works fine, but of course only for the active model.

Configuration:
 - EdgeTX 2.7 RC1
 - Eachine TX16S

raphaelcoeffic, 2022-04-02:
@RC-SOAR Mike, I'm having issues reproducing: it works for me. Could you please upload radio/model files so I can have a closer look? I tested with "SA" and also with "ON". Both work for me. Even with the touch sensor disabled, it will still "wake" the backlight, but not accept any input.

eshifri, 2022-04-02:
Works for me as well.

RC-SOAR, 2022-04-02:
Ah, my bad! The clue was in the model YAML file - noGlobalFunctions was set to '1', and sure enough global functions was disabled.

I've now enabled global functions and all is fine. My apologies!

closed 2022-04-02 by RC-SOAR

raphaelcoeffic, 2022-04-02:
> Ah, my bad! The clue was in the model YAML file - noGlobalFunctions was set to '1', and sure enough global functions was disabled.
> 
> I've now enabled global functions and all is fine. My apologies!

So you just looked it up in the YAML? Nice!

RC-SOAR, 2022-04-02:
Yep, YAML is sooo cool - it's great to be able to debug by looking at the source data  :-)