EdgeTX/edgetx #1791: Sticky mode stays active when a rapid set and reset occurs

thujones, 2022-04-02:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
During ground testing of GPS, rapidly flipped Prearm and Arm logical switch configuration to reset home position. Noted that unexpected circumstance of quad remaining armed despite configure Sticky disabler switch being in disarm position.

**To Reproduce**
Steps to reproduce the behavior:

- Using Radiomaster TX16S in this case, but I expect other radios would experience same issue

Setup:
1) Set logical switch L01:
     - Func AND
     - V1 SFdown
     - V2 SHdown
2) Set logical switch L02:
     - Func:  Stcky
     - V1: L01
     - V2: SFup

Process:
1) Pull SH momentary switch up (Prearm, if you will)
2) While holding SH in up position, pull 2 Position switch SF into up position (Arm + Prearm which is mapped to the mixer on a channel to ultimately set channel high where arm mode is set in betaflight)
3) While monitoring Logical switches, note that expected behavior of L01 going active by extension activates L02
4) Release momentary SH, noting that L01 becomes deactivated, and L02 stays "sticky" as expected
5) Go fly, then set SF to down position to disarm, this works as expected
6) Repeat steps 1-4 with exception of pulling 2 position SF and rapidly resetting it back to expected disarmed position (basically blip the arm switch SF from pulled to reset position quickly), noting however if done fast enough, L02 stays active (and by extension Betaflight will interpret this as actually armed)

Summary:
It appears during the transition of L02 going active from reading the value of L01, the Stcky condition does not recheck value of V2 fast enough to where mere humans can override it and get it stuck in an unexpected position

**Expected behavior**
Any time the V2 is set it should disable the sticky

**Screenshots**
My wife kindly recorded this issue happening, can be seen on unlisted Youtube video link at https://www.youtube.com/shorts/5NSznHcbvto

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

raphaelcoeffic, 2022-04-02:
I haven’t looked in great details into it, but please note there the logic switches are not evaluated so fast. I think it is approx. 100-200ms, so if you’re fast enough, this might happen. Please note that “normal” switches are evaluated faster, which might lead to discrepancies.

thujones, 2022-04-02:
Understood, and I am going to be mindful of this, just threw me for a loop when it happened! Thanks for confirming the likely cause, cheers!