EdgeTX/edgetx #1793: Companion simulator multi module protocol scan does not finish

elecpower, 2022-04-02:
**To Reproduce**
Either have the internal module set to MULTI or in the case of the attached log change OFF to MULTI in radio simulator.
Click on MDL button the MPM: Scanning protocols window is displayed but does not close.

In the debug window the Trace messages show that the scan does not stop at 100 attempts.
[FirmwareDebug_22-04-03_08-05-10.zip](https://github.com/EdgeTX/edgetx/files/8403453/FirmwareDebug_22-04-03_08-05-10.zip)

**Expected behaviour**
Should stop scanning at the limit though preferably for simulator it is pointless running at all.

**General information**
 - EdgeTX version 2.7.0 main
 - Simulated Hardware RM TX12S

eshifri, 2022-04-03:
The same behavior is observed in Windows as well.

raphaelcoeffic, 2022-04-03:
I believe this is `main`, and not `2.7` branch, right?

raphaelcoeffic, 2022-04-03:
Ok, the root issue is that the simulation does use FreeRTOS properly, so that the Software Timer task is not running (implemented by FreeRTOS itself). This would be solved by having the simu use the Posix/Windows port rather than trying to define it's own compatibility layer (see `rtos.h`) which is lacking quite stuff.

For now I will just disable the scanning, which does not make much sense here anyway.

eshifri, 2022-04-03:
I saw it on the main, but it was reported in 2.7 here: https://www.rcgroups.com/forums/showpost.php?p=48959165&postcount=2368

pfeerick, 2022-04-03:
Err, he said it was fine for 2.7.0, but was bugged when he used "EdgeTx_Companion_**2.8.0**-52473e1-x86_64.AppImage" ;) So the PR should fix that.

eshifri, 2022-04-04:
My bad. :-(

PR #1794 (2022-04-18): Use static MPM protolist on simu
This prevents the MPM protocol scanning to happen in the simulation, which would not work as the FreeRTOS Software Timers are not functional until we start to use FreeRTOS in simulation as well (Posix / Windows port).

Resolves #1793
commit f39f7a87 (2022-04-18): Use static MPM protolist on simu

Resolves #1793
closed 2022-04-18 by raphaelcoeffic