EdgeTX/edgetx #1795: When the [SYS Menu] adjusts the backlight, the string is not truncated correctly.

hmjack2008, 2022-04-03:
**Radiomaster TX16S**

- In Chinese translations (CN & TW), adjusting the [Backlight/Mode] on the [SYS Menu] page, the string was not truncated correctly.
[image]

- The English (EN) is correct.
[image]

- I tried adding a string terminator ('\0') and it displays correctly.
[image]

**\radio\src\translations\tw.h.txt**
#define LEN_VBLMODE    "\005"
#define TR_VBLMODE    "關閉\0" "按鍵\0" "操控\0" "全部\0" "開啟\0"

**General information**
 - EdgeTX version (TW and CN, 2.6 and nightly)
 - Hardware (TX16S)

pfeerick, 2022-04-03:
Thanks for catching that. The actual fault was in how the entries for that menu were being filtered so that "OFF" isn't shown on colorlcd... compiling the linked PR is back to showing only two characters for TW... does this look right to you?

[image]

[image]

hmjack2008, 2022-04-04:
Yes, that's what I need.
Thanks for the help!

PR #1798 (2022-04-05): fix: Use setAvailableHandler to filter blMode menu

Fixes #1795

Summary of changes:
- use `setAvailableHandler` rather than string processing which also mangled some translations
- this is colorlcd specific code, so doesn't need colorlocd conditional 

Malte, since this was your handywork... :)
closed 2022-04-05 by pfeerick