EdgeTX/edgetx #1796: Companion - can not set Trainer Mode to OFF

TheIsotopes, 2022-04-03:
**Describe the bug**
Trainer mode cannot be set to OFF in the Companion, but it works on the transmitter.

**To Reproduce**
using latest EdgeTX on Zorro

**Screenshots**

[image]

**General information**
 - EdgeTX latest 2.7 [092acd2](https://github.com/EdgeTX/edgetx/commit/092acd2d96955e2551292185631c3de391830206)
 - Zorro

elecpower, 2022-04-03:
Master/Multi also missing in Companion

TheIsotopes, 2022-04-04:
@elecpower I looked at it again on OpenTX, `OFF` is not displayed either there.
From my point of view switching it `OFF` is not intended, imo it shouldn't be in the list then. 

So only the Master/Multi entry is missing.

pfeerick, 2022-04-05:
This is not OpenTX :-P 

Companion should reflect the radios, which *do* have an off option now, and that is intended - Master is not the same as Off. 

All radios should have OFF as an option now, it is not restricted to being the NV14 only.

TheIsotopes, 2022-04-05:
ok then leave it in, I'm not mad about it either, find it even better
It just looked a bit unintentional since it's called `Off` and not `OFF` like everyone else.

PR #1806 (2022-04-05): Companion trainer modes OFF and Master-Multi available for all radios
Fixes #1796

Summary of changes:
- update trainer modes list item availability
- standardise trainer mode Off to OFF
closed 2022-04-05 by pfeerick