EdgeTX/edgetx #1802: GPS widget text colour

Flying-Potatoes, 2022-04-04:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)
 
Hi

I have a theme on my TX16S Max which makes it difficult to see black text in widgets. So I have set the text colour to white, but with the GPS widget colour defaults to black on power up. I have to go to widget settings, do absolutely nothing, and then out of settings to get the colour to white. Tried 2.7 rc1, and subsequent nighties. Thanks for all your work! 

Tia
Flying Potatoes.

pfeerick, 2022-04-05:
Can you give a little more information... which GPS widget is this? Or is this another widget such as the Value widget, set to read from the GPS sensor?

Flying-Potatoes, 2022-04-05:
Peter

Its the Moschotto OpenTX/EdgeTX GPS telemetry lua. I don't have my tx to hand right now, but will have later. 

Thanks.

pfeerick, 2022-04-05:
That's ok... that's what I needed to know. I'll have a look, and will either raise an issue or fix over there, as it's most likely a bug in the initalisation of his script... i.e. it's reverting to a default color, rather than the one configured in the widget options.

pfeerick, 2022-04-05:
It's a minor bug in the widget - the configured colour is only being processed when the widget options are updated... moschotto/OpenTX_GPS_Telemetry#24

closed 2022-04-05 by pfeerick

Flying-Potatoes, 2022-04-05:
Thank you very much indeed for your intervention. I saw what you were doing but wasn't able to respond till lunchtime. 

Best, 
David

Flying-Potatoes, 2022-04-05:
Peter

Thank you very much!

Best,
David.

Flying-Potatoes, 2022-04-05:
Peter

I feel slightly stupid, but am unable still to get this to work. I have
redownloaded the widget from releases with no joy. How do I get a version
with the merged patch?

Best,
David.

Flying-Potatoes, 2022-04-05:
Sorted via delete, copy, paste.

Sorry to be a nuisance,
David.

pfeerick, 2022-04-06:
Glad you got it working... that was exactly what I was going to recommend... happy flying! :)