EdgeTX/edgetx #1805: LUA - voice/timer synchronization

ZS-Man, 2022-04-05:
Hi,
until now I've been using on X10s with OpenTX and LUA F3K_Training script without any problem.
https://github.com/Mattoia90/F3K_Training

Now I switched to EdgeTX and I have two problems with this script:
1. slow and jerky movement when selecting a task (but that's not important)
2. after confirming any task, during 16s of preparation, the countdown on the display and the voice countdown are not synchronized. 

The voice countdown is slow. Same problem with countdown on flight end.

EdgeTX 2.6.0 and 2.7-nightly, in radio and simulator the same problem.

Is there any problem/difference in EdgeTX LUA implementation? Bug in EdgeTX or f3k training script?

Thank you.

JimB40, 2022-04-05:
Already mantioned to dev in Discord msg. It seems number wav files have some silence at the end that makes them longer than 1 second. As voice annoucements are queued (next is played when previous was finished) this makes de-sync between actual timer reading and what you hear.

ZS-Man, 2022-04-05:
Yes, you're right. The EdgeTX audio files have longer silences at the beginning and end (verified in audio editor). I copied the files from OpenTX and countdown in f3k_training works great.

Thank you for your advice

pfeerick, 2022-04-05:
Are these files from the release zips? As the raw audio in the sdcard-voices-repo is different to the voice pack release audio - it is  trimmed and volume normalised when the release zip is generated. If so, it obviously needs more tweaking.

mha1, 2022-04-05:
2.7 has a new LUA function flushAudio() which clears the audio queue. Try putting a flushAudio before playing the next .wav/number.

JimB40, 2022-04-05:
> 2.7 has a new LUA function flushAudio() which clears the audio queue. Try putting a flushAudio before playing the next .wav/number.

Good point. Still this won’t resolve problem as we talk about OS diriven timer.