EdgeTX/edgetx #1808: lcd.drawAnnulus display bugs using in widgets

JimB40, 2022-04-05:
Since I've started to use this func in Wirdgets Pack (timer & clock) I found it's behaviing oddly

During scroll Annulus is drawn on "other" main screen

[image]
[image]

Right position on timers
[image]

Ghosted on next main view
[image]

Also function ignores widget size and draws outside widget. Where other lcd.draw functionc are clipped correctly
[image]

**General information**
 - EdgeTX version 2.7.0 RC1
 - Hardware RM TX16S

jfrickmann, 2022-04-05:
When I found it in the code, I added some documentation, but I believe that the underlying drawing function was implemented in https://github.com/EdgeTX/libopenui/commit/df766a4da41a103dff983e7d52ff38f339bc37fa

pfeerick, 2022-04-06:
Yes, new/related to EdgeTX - PR https://github.com/EdgeTX/edgetx/pull/147

[image]

[image]

JimB40, 2022-04-08:
Hmm nice. not goinig to ask auhor :) 
So @raphaelcoeffic you just merged or have some idea how it works?

gagarinlg, 2022-08-18:
Do you know if this still happens with the current nightlies?

jfrickmann, 2022-08-19:
No, sorry!

pfeerick, 2022-08-20:
No, tested with today's nightly (e40e7ccc) on TX16S using the Clock widget from Robert's [FM2M Widget Pack](http://fm2m.jimb40.com/WidgetsPack.html) and this is good now (it had bad screen artefacts before). The only issue now is one of performance... i.e. as long as there is a widget on the active screen using `drawAnnulus`, overall radio performance seems to be sluggish. Yet you can switch to another screen that doesn't have a widget using `drawAnnulus` present, and all is good, until you switch back. Part of the "why is widget running when in settings?" question I think you raised before.

closed 2022-08-20 by pfeerick

gagarinlg, 2022-08-20:
Just take a look at the LVGL code to draw this stuff and you will understand the speed. ☹️