EdgeTX/edgetx #1809: Creating a new model via wizard not worky

pfeerick, 2022-04-06:
**Describe the bug**
When using creating a new model, and choose any of the "Wizard" options, the wizard doesn't appear at first, and then when you exit the model select screen you can see it but can't interact with it. 

**Expected behavior**
Wizard should be in focus ;)

**To Reproduce**
https://user-images.githubusercontent.com/5500713/161927926-30b4864d-3675-4c28-9dcf-f1d0a54b0f85.mp4

**General information**
 - EdgeTX version: 2.7.0-rc2
 - Hardware: RM TX16, simulator

jfrickmann, 2022-04-06:
Hmm, strange as it did work before. Did anything change in libopenui?

Here are the lines of code involved:
https://github.com/EdgeTX/edgetx/blob/d9b51f8ecf0f14b45e4f11e5d2f9f2637716a219/radio/src/gui/colorlcd/model_select.cpp#L165-L175

The general idea is that I first remove the two pages for selecting templates by calling `deleteLater` on them, and then I find the model select page with `focusWindow` and attach the Lua Standalone window to that. 

I thought that I had found something that worked, but evidently it did not work for long! So maybe someone with more libopenui savviness can help out? @gagarinlg do you have any suggestions?

pfeerick, 2022-04-06:
There havn't been any changes to libopenui at all ... we've not moved the pointer for it at all this cycle AFAIK... and yeah, that's what is puzzling me... It was working perfectly on merge of the template PR... I can't place anything that changed that should have done it... i.e. yes, I added the `#ifdef`, but would kill it entirely...

raphaelcoeffic, 2022-04-07:
I would need to have a close look, but I have the feeling this should be done differently. I am not sure `deleteLater()` will handle the focus changing in place.

jfrickmann, 2022-04-11:
I have tried a bunch of things now without success. Then I tried git bisect. They were all bad. Then I checked out v2.7.0-rc1 and built. It was also bad. At this point, I have no clue why it ever worked, and I am probably getting to the end of the line for me ...

jfrickmann, 2022-04-12:
I think that I have a clue. After a lot of bisecting, it appears to be related to #1709, but weirdly, I think that it requires `make clean` for a complete rebuild to show.

In that PR, I moved updating of the model button (name, call `update`) from before calling `deleteLater` on the two template pages and attaching `StandaloneLuaWindow` to the destructor of the first template page. So I am guessing that the model button runs `update` after the `StandaloneLuaWindow` was attached, and that somehow messes up focus. 

I will try if I can come up with a fix tonight.

gagarinlg, 2022-04-12:
Focus is a mess with libopenui ☹️

jfrickmann, 2022-04-12:
Yeah it's odd how a background window can steal focus from the visible window ...

Is any of that going to change with LVGL?

closed 2022-04-14 by pfeerick