EdgeTX/edgetx #1813: radiomaster t16s

pds59, 2022-04-07:
i update via buddy to latest version 2.7 rc2 ,process was ok unplug usb cable the radio does not start, 6 position blinking but the radio will not power up. i try to reload software via buddy but i get message Could not erase existing firmware Device firmware may be read protected, preventing

odalem, 2022-04-08:
I also had and issue with 2.7 rc2 on the T16S. I tried upgrading by putting the firmware on the SD card then flashing the boot loader and then rebooting, attempting to reboot into the 2.7 boot loader was unsuccessful the radio just started normally, I had to revert to 2.6. I believe there's a problem.

pds59, 2022-04-08:
Tks but the radio Is totally look and i have no chance to operate i Will try to put old firmware on sd  and seePaolo de scaglia

rotorman, 2022-04-08:
See https://github.com/EdgeTX/edgetx/wiki/Unbrick-your-radio

pds59, 2022-04-10:
Solved back to life again special super tks for your help tks again

 

Habitatika di Paolo de Scaglia

Via del Salice 1 

28836 Vezzo VB

Cell. +39-3466393930

 

BV6BANCNFSM5S2BU4UQ> .
<https://github.com/notifications/beacon/AUTY7X7UOSGW65F7D3BTBPDVEBV6BA5CNFS
M5S2BU4U2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOIET5BUI

closed 2022-04-10 by rotorman