EdgeTX/edgetx #1823: [colorlcd] Next/Prev page button stops working after entering Modules/RX version in info tab

obeny, 2022-04-11:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
1. Select a model that has external TX configured. I used a model that has ELRS TX using CRSF protocol.
2. Go to SYS->Info (version)
3. Enter Modules/RX version
4. Press any of page+/page- when version view is visible, switch to the other tab, e.g. settings.
5. Exit Module/RX version preview window

All of the buttons: MDL, SYS Page+/-, RET are not working, radio reboot is required.

**Expected behavior**
Buttons should work after exitting the Module/RX version window.
It is more reasonable to block using page+/page- buttons to stay on the same tab while checking version info.

**General information**
 - EdgeTX 2.7.0-rc1, rc2
 - RadioMaster TX16S
 - HappyModel ELRS 2.4 TX (CRSF protocol)
 - Radio setttings migrated from 2.6.

Can you confirm that you have same behavior?

pfeerick, 2022-04-11:
Ah, bugger... forgot about #1582 ... yes, it's repeatable, and at worst, can trigger a EM due to focus being lost.

obeny, 2022-04-11:
Oh, I'm sorry folks, used search but couldn't find similar bug, so created new one.

pfeerick, 2022-04-12:
No problem... when you opened this, it set off the reminder bell. But it can sometimes be a problem though... trying to pin down that issue even if you are sure it has been mentioned before... I'm sure I'm guilty of opening an issue or two simply because I couldn't find the prior mention...

closed 2022-04-27 by pfeerick