EdgeTX/edgetx #1824: Jumper T16 2.7 rc-2

Stinky80, 2022-04-12:
after flashing the latest version "2.7 rc-2"  the radio does not boot ... black screen ... but the flashing entry mode works (T4 + T1 + Power)
you can safely return to 2.6

pfeerick, 2022-04-12:
Would you be able to you try 2.7rc2 again, but rename (don't delete!!) your MODELS and RADIO folders... and see if it boots then. If it does, can you then try turning the MPM / MULTI on, and see if the radio will power cycle successfully. I am trying to rule out some weird form of settings corruption, or if the MPM needs an even longer delay after being turned on to prevent the power supply from browning out. 

If that doesn't work, rename your folders back so your settings are restored, and roll back to 2.6 for now...

Stinky80, 2022-04-12:
As promised, I answer on the test. renaming the model and radio folders had no effect. The problem remained ... with the firmware back to 2.6 without changing the folder names, the radio said that the data from the SD card was incorrect, but it started up. 2.7 rc-1 2.7-rc2 do not work on jumper t16 board version 2 nightly build released after rc-2 with tag 2.8 started normally

Stinky80, 2022-04-12:
*night build 2.8 works

Sorry  google translate :(

pfeerick, 2022-04-13:
That is strange as there is nothing *significantly* different between `main` (nightlies) and 2.7.0-rc2, nor is there anything specific to the Jumper T16 recently. We may just have to mark it up as a potential known issue for 2.7.0. If the currently nightly is working for you, there should be no harm continuing to use that. Perhaps try again after 2.7.0 goes to release, and if it still doesn't work, try the latest nightly, as that will have all of the 2.7.0 fixes and seems to be working properly for you.

closed 2022-04-20 by Stinky80

pfeerick, 2022-04-20:
Hm... please don't hestitate to mention if it plays up again... as I suspect it will in a future release until we pin down the root cause.

Stinky80, 2022-04-22:
of course I will follow the work of the releases, I can try the latest releases for performance .. I also want to learn how to assemble the firmware with the flags I need to try to understand the operation of the touchscreen, gyroscope and GPS for the T16 jumper - this is very interesting .. in using bluetooth I was disappointed .. crooked and does not make sense ...

Stinky80, 2022-04-22:
где можно понять какие флаги сборки описаны и где их искать?  я только учусь разбираться с кодом

rotorman, 2022-04-22:
English only please. You are asking about build flags, see: https://github.com/EdgeTX/edgetx/wiki/Compilation-options